PiSupply/PiJuice #799: How to fully reboot the pi + its hats?

OhSoGood, 2021-12-07:
Hi,
I would like to fully reboot the pi and its hats, notably when a hat goes wrong and needs a reset. A reboot in linux only restart the pi, not its hats. Can we do it with the pijuice and how? Could you give some shell or python examples?

Thank you!

tvoverbeek, 2021-12-07:
How do you reset your hat(s). Is power cycle (i.e. cycle 3.3 and 5.0V GPIO)?

OhSoGood, 2021-12-08:
Yes, by power cycling

tvoverbeek, 2021-12-08:
Can be done if your Pi + HATs can run on only battery power from the PiJuice (external power only connected to PiJuice)
Executing SYS_FUNC_HALT_POW_OFF (i.e. when connected to a SWx) will shut down the Pi and then remove the 5V
to the PI after a delay (ca 30 sec).
Then wake up the Pi again with a button press.
See e.g. https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-config-buttons-menu
With external power connected to the Pi it will not work.

Note: For a Pi4 you can program the bootloader to switch off all power buses when shutting down. This will reset all the USBs and also switch off the 3.3V. However the 5V stays on. See the official RPi docs for details. Do not know if removing the 3.3V only will reset your HATs.

OhSoGood, 2021-12-09:
Actually, I wanted the ups, the pi and the hats to automatically restart after the power off.
Here is how I do in python:

```
#!/usr/bin/python3
import os
from pijuice import PiJuice

pijuice = PiJuice(1, 0x14) # Instantiate PiJuice interface object

# By setting WakeUpOnCharge to 0, the Raspberry will start as soon as power is back
pijuice.power.SetWakeUpOnCharge(0.0)

# Set power off time-out to 15 seconds (or shorter if your Pi is shutting down faster)
pijuice.power.SetPowerOff(15)
# Now shutdown
os.system("sudo poweroff")

```