PiSupply/PiJuice #800: Battery Drain issue with PiJuice even when power is connected

rdragonrydr, 2021-12-11:
I have a crappy USB cable or adapter - not sure which - and if I'm powering the system via the Raspberry Pi's USB port, the pi boots up off of the USB supply, then wakes the hat. At this point the hat is now powered on and detects the low voltage on the GPIO pins (presumably; this happens before the pi finishes booting, so I'm not able to view the stats). The hat then switches on its own supply and starts providing additional power to bring the voltage up.

The end result is a dead battery and a powered-down pi thanks to the low-battery trigger despite having power the whole time.

Is it possible to configure or update the firmware so that it can better detect external power and taper down its output to see if the supply is capable of running the pi on its own? Right now it seems a bit insensitive.

tvoverbeek, 2021-12-11:
Have you used the 'NO POWER' event?
The pijuice service polls both power inputs (PiJuice usb and PiJuice GPIO) and considers there to be no power
from the GPIO if it is not present or bad.
What does the PiJuice Status tab say ('Battery' and 'GPIO power input' lines) for this situation?

rdragonrydr, 2021-12-11:
I have not used the 'no power' event. I don't want to actually shut the pi down since it _has_ power but isn't being detected for some reason.

I had to check this again to be sure...

I got 4.7V on the GPIO pins but it was displaying 0 amps this time in the hat status. Convenient, the hat didn't auto-power on this time!

I pressed the power button to wake up the hat (normally it _does_ do that on its own - note though that now that it's on I don't think I can shut it down short of turning off the host pi too?) and the voltage is now 5.028V and between 0.4 and 0.7 amps **from** the hat on idle.

It says GPIO is not present. Battery is 96%, 3.994V, 25 C, normal. Micro USB is not present (I'm attempting to power it from the pi at the moment).

tvoverbeek, 2021-12-11:
Is the Pi running (which Pi type?)?
4.7 V on the 5V rail is definitely too low. You should get an undervoltage warning on the Pi (the lightning bolt on the desk top).
0 Amps means there is no current flowing either to or from the Pi (5V produced by the PiJuice is off).
Did it say GPIO INPUT BAD or something else (WEAK/NOT PRESENT/PRESENT)?
Hat must be on, since you can query the status.

The power button on the PiJuice turns on the 5V from the PiJuice to the Pi via the GPIO.
Hence you get the proper 5.028V to the Pi (lightning bolt should have disappeared).
Battery parameters look good.

==> Dump your bad power supply.

rdragonrydr, 2021-12-11:
Yes, it is running. A raspberry pi 2b
Yes, it does do that.
I'm aware it was off at the time. Of course, turning it on means that it takes over...
It said "not present" as noted in my prior comment.
I'm not sure of the details. I could query it and it was lit up solid green, so it was presumably on/awake, but its power output wasn't on that time for some reason. Normally it powers on and starts the output too.

It might be interesting to have a way to turn the hat off even with the pi on (or at least turn off power output) if you don't currently need the UPS.

Yeah, it looks like I need to. Still, this is something it'd be neat to have a workaround for.