PiSupply/PiJuice #801: Incorrect "Charge Status" Led Issues

rdragonrydr, 2021-12-11:
For the fun of it since I don't use the "user" LED, I set both LEDs to display charge level. This works properly for >50% charge, but for battery <50% and charging, LED1 displays only red and not the blinking blue. D2 works as expected.

Also, even in its original LED configuration (D1 to charge status, D2 to user), the LEDs never show yellow for the ">15, <50" status as described in the documentation. It just goes straight to red the moment it drops below 50%. It also does not show yellow for the two-LED setup either. I have _not_ tested to see if D2 **on its own** shows yellow at this power level, for full disclosure.

tvoverbeek, 2021-12-11:
What are your settings for R, G, and B for both D1 and D2? This influences the display of the charge level.
The 'never show yellow' is a known problem. Yellow is made from the combination of R and G.
With the default settings and the type of Led used red overshadows the green so you do not see
a proper yellow. I personally see only red below 50% charge.

rdragonrydr, 2021-12-11:
I had all three set to 80. (I've been trying to figure out - is that out of 100 or 255?) I can't see the green pixel on at all inside the casing when I look close, though, even though when the blue is on for the charge indication I can clearly see both blue and green.

rdragonrydr, 2021-12-24:
Are those good or bad settings? What is the suggested value for if I want to see the yellow color for mid-battery readings?

tvoverbeek, 2021-12-24:
Values can be specified in the range 0-255.
Try out various combinations of R and G till you find something yellowish.
Set these as values for the charge level led.
Between 50 and 100% only the R value is used.
Between 15 and 50% both the R and G values are used.
Below 15% only the R value.
B is only used as charging indication

rdragonrydr, 2022-01-26:
Settings that worked on my unit to show the yellow <50% state are:
R: 50
G: 65
B: 60

closed 2022-01-26 by rdragonrydr