PiSupply/PiJuice #802: Log file?

icsy7867, 2021-12-14:
I am getting random reboots on my pi4, and I am not sure if it's from the pijuice.

Is there a log file for the pijuice I can peruse to see what it is triggering?

tvoverbeek, 2021-12-14:
Sorry, no log directly from PiJuice available.

icsy7867, 2021-12-14:
Thanks for the prompt reply!  I'll keep digging through the system logs.

Also I really like your product.  I had to have a long trip due to a family emergency.  I have a car hotspot and streaming server.  The pijuice gives me a great way to safely shutdown when the car is turned off (no need to hardwire into the battery and acc power!)

mmilann, 2021-12-14:
@icsy7867 you can try latest firmware [V1.6 development ](https://github.com/PiSupply/PiJuice/tree/master/Firmware/Sources_dev).
It supports kind of logging. Attached
[PiJuice-V1.6_2021_09_10.elf.binary.zip](https://github.com/PiSupply/PiJuice/files/7712949/PiJuice-V1.6_2021_09_10.elf.binary.zip)
 is the build that you can write using pijuiceboot tool (need to use [updated version](https://github.com/PiSupply/PiJuice/blob/master/Firmware/pijuiceboot) also).

PiJuice logs events like wakeup or 5V_regulator_off and it is saved in mcu memory. There is python script you can use to read logs [pijuice_log.py](https://github.com/PiSupply/PiJuice/blob/master/Software/Test/pijuice_log.py). Before test you need to enable log with command:
python3 pijuice_log.py --enable "ALL|5VREG_ON|5VREG_OFF|WAKEUP_EVT|ALARM_EVT|MCU_RESET"

After you can just read log with command:
python3 pijuice_log.py
or read to file:
python3 pijuice_log.py ./pijuice_log.txt