PiSupply/PiJuice #803: battery profile, location

GekoCH, 2021-12-23:
can someone tell me where the battery profile settings are stored?
Can't find it in the `/var/lib/pijuice/pijuice_config.JSON` file or when I dump the config file with:
`python pijuice_util.py --dump > dump.js`

I need to copy those Infos so I can deploy them to other pijuice devices...

tvoverbeek, 2021-12-23:
They are in the firmware source: https://github.com/PiSupply/PiJuice/blob/master/Firmware/Sources-V1.5_2021_02_06/Inc/battery_profiles.h

GekoCH, 2021-12-23:
So the custom ones are also stored into this file?

tvoverbeek, 2021-12-23:
The custom profile in the file is the initial setting. If you change it for a custom battery (e.g. some 18650) it replaces the custom profile in the firmware on that particular PiJuice.
If you need it on other PiJuices you will need to change it on all of them.
The other profiles are the same on all PiJuices provided they use the same firmware version.

closed 2021-12-27 by GekoCH