PiSupply/PiJuice #807: Behavior of System task options

Jan36Nikon, 2022-01-12:
Hi everyone,

I want to ask about behavior of System task options:

Wake up on charge:
Minimum charge: 

I set "wake up on charge" at 90% but when i charged battery from 70% to 100% nothing happend.
Charging was done when device was in power off mode waiting to switch itself on on the next day.
(Switching on and off is working correctly)

I set "minimum charge" at 20% and when battery was under 20% PiJuice start at every switching on sequence and after few secconds it switch itself off. That is OK for me.
What will happen when device will start with 21% battery and before it finished its work for that power on sequence battery will decrease to 20%? It will switch off the device even before if finished its work?

Thanks a lot for clarifying this issue.
Have a nice day,
Jan

[image]

tvoverbeek, 2022-01-12:
Wakeup on charge
From your screenshot it looks like you are not running the latest software and firmware version.
In older versions you have to set 'Wakeup on charge' just beforen you shutdown the Pi.
In the older versions the setting gets cleared when the RPi starts again.
This would explain why nothing happens in your case.
In the latest version (package version 1.7, firmware version 1.5) there is an option to store the 'Wakeup on charge'
setting in the firmware non-volatile memory.

Minimum charge:
The min charge shutdown is performed by the pijuice service running on the RPi.
So the Pi has to boot in order to get the service running even if this gets the charge level below 20%.
This takes a few seconds, which is what you are seeing.

See also https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/changelog

Jan36Nikon, 2022-01-12:
Thank You for fast reply.
I will check firmware version because screen is actually from PiJuice manual not my screen.

And do you think that when device will start with 21% battery and will decrease to 20%? It will switch off the device even before it finished its work immediately, or it will shut down for example after 1 minutes or something like that?

Thanks, Jan

tvoverbeek, 2022-01-12:
When the device starts at 21% it will shut down as soon as the pijuice service detects the charge level is below 20%.
The shut down is an orderly shut down.
Of course if you start other work which takes longer, this will be interrupted in the same way as when you start a manual shut down during the work.

Jan36Nikon, 2022-01-13:
So in that case it will not set another wakeup and it will wait for reach of 90% of battery by charging and device will stars and set another wakeups.
Ok that is acceptable.

According to
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/changelog
When I check Wake-up on charge restore It should start at 90%. Perfect.

My question is on that part:
"wake-up on charge will be automatically activated when charging stops after power source is removed."

What will happen when solar panel will be connected during cloudy day?
Will it randomly start up?

Thanks,
Jan

tvoverbeek, 2022-01-13:
It will start up when the battery goes over 90%. When it will shut down depends on your other settings (i.e. low charge). It will *not* shut down when the charge goes below 905%.
Solar panels will still deliver some power even when it is cloudy. So the 'power source removed' usually does not apply when a cloud moves in front of the sun.

Jan36Nikon, 2022-01-13:
Yes, thank you for your info.
Great help.
Jan