PiSupply/PiJuice #808: Using external (PiJuice 12000) battery with PiJuice (non-Zero)?

jimklimov, 2022-01-21:
The web-site and docs seem to suggest that the PiJuice can be used with an external power sources, such as Solar or Wind generators, presumably those would charge its batteries and let it last longer when off-grid.

The site also suggests e.g. in https://uk.pi-supply.com/products/pijuice-standard page that PiJuice batteries are a nice match, same level as solar strips. The README.md also says:

> That's why we have provided means to support bigger battery sizes like 5000 or 10,000 mAH+ to help your PiJuice last up to 24 hrs +.

But just after that, discussing the backup power while changing a battery, it casually mentions frying the device:

> You could for example even use a battery bank whilst replacing either the onboard battery or one connected to the screw terminal (you shouldn't connect a battery on board and to screw terminal at same time though or you will fry your PiJuice).

This startled me as I was just about to piece together the bits I have in a box, and suddenly in the middle of nowhere found this line while looking for some other instructions and inspirations.

So which one is it? Can the PiJuice use *both* the provided 1820mAh battery *and* the 12000mAh one through the four big screws? Or should at most one be connected at any time, otherwise things burn? Are there similar caveats about actual renewable sources connected to that terminal?
* If there can be only one, I think the notice should be much more prominent, including the web-site/shop overview pages like "You can use one of these larger batteries, but be sure to detach the small one first!"

And can the bigger PiJuice HAT use the "PiJuice 12000 battery" at all actually, or is it really meant for the PiJuice Zero pHAT (with hardwired white plug and all)?

On a similar note, I do also have a PiJuice Zero pHAT but not an actual PiJuice Zero computer; can the pHAT (with the external larger battery) be used with the Raspberry Pi 4B instead of PiJuice? Can both be attached as a stack (including two batteries this way, one for each HAT) - or would they conflict electrically and/or programmatically?

Finally, kudos for providing the additional power port (and the non USB-C option to feed the Pi 4B at that), because I have an angled converter from microHDMI to big HDMI cable that my display has, and it overlaps in space with the USB-C port on the 4B so I can't actually use the display :) Maybe this bonus point should be highlighted, especially for shipping multi-component kits (e.g. "put hardware together first, then power up from convenient source and install the software").

tvoverbeek, 2022-01-21:
The external power source has to be connected to the PiJuice micro-USB or to the J4 connector. Both are in parallel and allow input 4.2 - 10V. 
When replacing the PiJuice battery you could use a battery bank connected to J4 or to the micro-USB to power the PiJuice/Rpi (or just connect a 5V power supply to the PiJuice micro-USB).

From the description (only a single cell battery to be used) it is clear that you should not connect 2 different batteries to the PiJuice battery connector.
You can use either the 1820mAh or the 12000mAh battery. To connect the 12000mAh battery to the screw terminals of the HAT you can cut off the battery JST-connector. See e.g. https://learn.pi-supply.com/make/how-to-setup-connect-your-pijuice-battery.

The PiJuice Zero HAT can be used with any 40-pin GPIO RPi. However on the 4B for example you will need a GPIO extender like https://www.adafruit.com/product/1992. Of course you cannot use the pHAT mounting holes above the middle of the larger 4B board.

Note that if you power th 4B from the PiJuice via the 5V GPIO there is a limit on the power the PiJuice can supply to the 4B. For normal use it is ok, but with power hungry USB peripherals it might not be sufficient. 3A is definitely too much.

jimklimov, 2022-01-21:
Oh, right. I took the 1820mAh battery off now to take a look around, and the screwable connector holes are actually blocked by the battery body. Smart, if there can be only one (sparks and lightnings would be suitable) ;)

tvoverbeek, 2022-01-21:
@jimklimov This is even documented in https://github.com/PiSupply/PiJuice/tree/master/Hardware#top-of-the-board
Quote:
```
The screw terminal is intentionally facing inwards, towards the onboard battery, for safety...this is not a
mistake! The idea is to make it harder for you to accidentally plug in two batteries at once and fry your
PiJuice.If you would like to turn the screwterminal to face the other direction you can do so by unscrewing
all four screws, turning it around, and then screwing them back in again. Please note that you do this at
your own risk and it will void your warranty.
```

jimklimov, 2022-01-21:
Well, at least it does make it hard to make an inadvertent mistake. But I still propose that highlighting this more visibly would be useful (for shopping also).

Similarly, is there a frying danger if one mixes the PiJuice battery and a "renewable source" through same terminals? 
What if USB power is also plugged in (so 3 things at once)?