PiSupply/PiJuice #810: I2C Command Reference

AlexZatti, 2022-01-25:
Hi all, 
I'm deployng my application in GO in order to monitor the status.

Do you have a description of command and bit mapping of response?

At the moment I have doubt about the Status (Command 0x40), Battery Temperature (0x47) and battery voltage (0x49).

Thanks 
Ale

tvoverbeek, 2022-01-25:
There is no low level I2C command/response document.
You will have to deduce it from pijuice.py (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py)

For example for the status command (0x40).
There is one data byte returned (+ checksum)
```
-------------------------
|b7|b6|b5|b4|b3|b2|b1|b0|
|  |  |  |  |  |  |  |  |
-------------------------

b0 isFault status
b1 isButton status
b3,b2 battery status:
  00  'NORMAL'
  01  'CHARGING_FROM_IN'                        
  10  'CHARGING_FROM_5V_IO'
  11  'NOT_PRESENT'
b5,b4 powerInput status
b7.b6 power Input5vIo status
  00  'NOT_PRESENT'
  01  'BAD'
  10  'WEAK'
  11  'PRESENT'
```
See the GetStatus() function in pijuice.py (lines 165-180)

For Battery temperature (0x47)
1 byte is returned to be interpreted as a signed 8 bit number (-128 .. 127)
GetBatteryTemperature() function lines 258 .. 267

For Battery voltage (0x49)
2 bytes are retuned to be interpreted as a 16-bit number.
Scaling is in mV
GetBatteryVoltage lines 269-275
Scaling is done in pijuice-cli/pijuice_gui

AlexZatti, 2022-01-25:
Thanks! I'm doing a reverse engine of the source you told me...

Querying the 0x40 (the status) I don't why my  "b0 isFault status" is always one!

Checking PiJuice_cli/status "isFault" is zero. 

All the other one apper fine. 

Ale

tvoverbeek, 2022-01-25:
pijuice_cli does not look at the status bit b0, but uses pijuice.status.GetFaultStatus().
So if you want you can ignore b0 and use GetFaultStatus() instead.

vinthewrench, 2022-02-01:
it it helps, I am working on a similar issue for one of my home projects,  I have some C++ code here that gets the voltage and status..  I'll package it up latter, but for now this works -- 
https://github.com/vinthewrench/chickencoop/blob/main/coopserver/CoopMgr/PiJuice.cpp
https://github.com/vinthewrench/chickencoop/blob/main/coopserver/CoopMgr/PiJuice.hpp
https://github.com/vinthewrench/chickencoop/blob/main/coopserver/CoopMgr/I2C.cpp
https://github.com/vinthewrench/chickencoop/blob/main/coopserver/CoopMgr/I2C.hpp

AlexZatti, 2022-02-03:
Thanks all! 
I more or less converted the Read Status in Go and it's working. At this point What is the meaning of the bit0 of the GetStatus 0x40  command? The GetFaultStatus doesn't report error. Is it deprecated?

Ale