PiSupply/PiJuice #811: Unable to install via terminal on Raspbian

K9LOL, 2022-01-30:
I’m trying to install the software using the commands listed in the documentation but am receiving an error “Unable to locate package” for both pijuice-gui and pijuice-cli. I ran apt-get update first just to be sure, but it didn’t help. Is there a new way to get the software via terminal? Thanks.

tvoverbeek, 2022-01-30:
I suppose you are on Bullseye. The pijuice packages are not (yet) in the Bullseye repository, which is why you were not able to install them.
Download the packages (pijuice-base_1.7_all.deb and pijuice-gui_1.7_all.deb) from https://github.com/PiSupply/PiJuice/tree/master/Software/Install.
You can then manually install them from the directory where you downloaded them:
```
sudo apt-get install ./pijuice-base_1.7_all.deb 
sudo apt-get install ./pijuice-gui_1.7_all.deb 
```
The battery status icon in pijuice gui will have a black background in Bullseye. Will be fixed when the packages for Bullseye are ready.

K9LOL, 2022-02-02:
My apologies, I’m new to the Pi and Linux and wasn’t sure what Bullseye was. I followed your directions and it installed, but I saw in the output that it apparently is in the Bullseye now, so thanks for that!

closed 2022-02-02 by K9LOL