PiSupply/PiJuice #813: How to script that (instead of using pijuice_cli)?

OhSoGood, 2022-02-02:
Hi,

I have a couple of pijuice to setup, and sometimes there are away not connected to internet, so I would need to set up their pijuice "profile" by code (through a python or shell startup script in the Raspberry). I've tried to looked through pijuice_cli code, but it's so dense...

What I would like to do by code is:
- set battery profile to the 12000mAh battery
- in "general" tab, check "No battery turn on"
- in "system task" tab, check "System task enabled", check "Wakeup on charge" (+ Trigger level to 15% and check Restore), check "Min charge" (and set threshold to 8%).

Could someone be kind enough to show me the API to use? A sample script would be warmly welcomed!
Thanks

tvoverbeek, 2022-02-02:
Here a script (not tested):
```
from pijuice import PiJuice
pj=PiJuice(1,0x14)
#
# Set battery profile to the 12000mAh battery
pj.config.SetBatteryProfile('PJLIPO_12000')
#
# Set "No battery turn on"
ret=pj.config.GetPowerInputsConfig()
if ret['data']=='NO_ERROR':
   conf = ret['data']
   nv = ret['non_volatile']
   conf['no_battery_turn_on'] = True
   pj.SetPowerInputsConfig(conf, 'non_volatile'=nv)
#
# Set "Wakeup on charge" at 15% and select Restore
pj.power.SetWakeUpOnCharge(15, 'non_volatile'=True)
```
The other settings (System Task Enable and MinCharge) are in the JSON settings file
See https://github.com/PiSupply/PiJuice/tree/master/Software#json-configuration-file for details.

OhSoGood, 2022-02-02:
Thank you a lot.
On my read-only system booted with argument "ro systemd.volatile=state", the folder /var/lib/pijuice doesn't exist after boot. Have you ever faced this case? How to bypass that? Is this the pijuice unit file which coud be incomplete (no RuntimeDirectory , but only WorkingDirectory ) ? I don't use such notions that often, so I may be wrong.

tvoverbeek, 2022-02-04:
Your "systemd.volatile=state" mount /var/lib (system state directory) as a temporary filesystem in ram. This means it is empty after each boot. So somehow you will have to recreate /var/lib/pijuice/pijuice_config.JSON after each boot before starting the pijuice service.