PiSupply/PiJuice #814: Some questions about the PiJuice

clarke-baxter, 2022-02-03:
My project uses a Raspberry Pi 4B with an official Pi wall wart power supply.

The GPIO pins are accessible to connect wires, but I have insufficient headroom to mount another board directly to the Pi. 

Can I connect the PiJuice without mounting it directly and if so, is there any loss of functionality?

It is important that I can do a safe shutdown of the Pi followed by a complete shutdown of the PiJuice with a switch.

I need to be able to power-off and power-on the Pi4 and the PiJuice with a single externally mounted switch. Ideally this would be a momentary switch with built-in LED indicator. Is this possible, and if so do you sell a suitable switch?

tvoverbeek, 2022-02-04:
> Can I connect the PiJuice without mounting it directly and if so, is there any loss of functionality?
Yes, you can. PiJuice only needs the I2C pins and power and ground.
See https://pinout.xyz/pinout/pijuice

If you want to use a single external switch to both power on and off, I suggest to connect the external (switched) power to the Pi4 and program the PiJuice to shutdown safely and remove the 5V to the Pi (SYS_FUNC_HALT_POW_OFF) when it detects 'No Power'. The actual shutdown will then be performed on battery power.
The PiJuice main function in this scenario is as UPS.

clarke-baxter, 2022-02-04:
OK, got it. Thank you! Order placed.

tvoverbeek, 2022-02-04:
Closing the issue. If you do not agree you can reopen ite.

closed 2022-02-04 by tvoverbeek

clarke-baxter, 2022-02-05:
Apologies for re-opening this, but I have realised I did not ask the question properly.
Basically I am looking for a laptop-style power arrangement. 
When the power switch is pressed the Pi come alive.
When there is no external power the Pi runs on the battery. 
When the power is plugged in the Pi needs to continue uninterrupted and the battery needs to charge. 
When the power supply is unplugged the Pi continues on battery power.
When the power switch is pressed the Pi and the PiJuice power down.

Is this possible?

Thank you.