PiSupply/PiJuice #815: PiJuice Zero sleep-mode

Ohlhues, 2022-02-04:
Dear All,
I just bought a PiJuice Zero for a research project. In my project, power consumption is very important. As I can read, the module uses 0.5Amp in sleep mode ?? Is that right? Can I reduce power consumption further? 

Yours sincerely
Anders

tvoverbeek, 2022-02-04:
No that is not right. When the PiJuice is in low power mode it uses ca 1mA. It can survive weeks in low power mode when on battery only.
Where did you find the 0.5A?

Ohlhues, 2022-02-04:
Dear tvoverbeek,
First I must inform you that I think your product is super cool. I read about power consumption via this link: https://github.com/PiSupply/PiJuice/issues/794

I do not measure the power so I do not know how much power my module uses. My battery dropped 10% overnight. I use HARD_FUNC_POWER_OFF. I think it's my battery profile. I bought your 2500 mA LiPo PIS-0379. I read that you encourage using the 2300mA profile. But is that completely true? 

Sincerely, Anders

tvoverbeek, 2022-02-04:
Issue #794 says 0.5 mA, not 0.5A.
How do you use HARD_FUNC_POWER_OFF with SW1?
A long press 1 initiates a normal shutdown but the default setting does not remove the 5V to the Pi.
Does the red power led on the Pi remain on? (means the 5V from the PiJuice to the Pi is not switched off)
The long press 2 (Longer than logn press 1) does a hard power off, but does not do an orderly shutdown.
When the 5V from the PiJuice is really off the Pi red power led should be off and the PiJuice leds should blink very slowly and with low intensity.

If your firmware is up-to-date then you should be able to select the PJLIPO_2500 profile.

Ohlhues, 2022-02-05:
Hi Tvoverbeek,
Now I've found out what I did wrong. I shut down my RPI Zero 2 with HARD_FUNC_POWER_OFF. Last night I tried to shut down with SYS_FUNC_HALT_POW_OFF and now it works perfectly. Zero% battery used in one night :-) As you mention, it is 5 Volts that use power. I have also upgraded the firmware and now I have the 2500 mA profile. Super :-) Thanks for your help. 

Sincerely, Anders

tvoverbeek, 2022-02-05:
Glad you sorted it out.
I am assuming you are svensk or norsk.
Lycka till

Ohlhues, 2022-02-05:
Close :-) I'm from Denmark. Do you come from Scandinavia yourself?

Best,
Anders

tvoverbeek, 2022-02-05:
@Ohlhues No, I am from the Netherlands but lived 7 years in Sweden (studied and got a doctors degree at Liköpings Universitet)

tvoverbeek, 2022-03-22:
I am only a volunteer not employed by pi-supply. Contact Pi-Supply/Nebra directly (e.g. sales@pi-supply.com) with this question.