PiSupply/PiJuice #817: /usr/bin/pijuice_cli: No such file or directory

virtualarchitectures, 2022-02-06:
Hi, I'm trying to set up the PiJuice headless. I used the fix by @tvoverbeek for issue #811 to get around the installation issue. Now I'm unable to run the `pijuice_cli` or `pijuice_cli.py` application to configure the hat. When I run those commands the terminal returns `/usr/bin/pijuice_cli: No such file or directory`. However, if use WinSCP to inspect the device file system I can see the files are present (see image). I tried rebooting my device. I also checked the Python version on the device which is Python 3.9.2.
[image]

tvoverbeek, 2022-02-07:
Files look ok on your screenshot. Which OS are you using? 32-bit or 64-bit?
The pijuice_cli in the 1.7 package does not run on 64-bit since it is a 32-bit executable.
If you want to execute pijuice_cli.py directly use the following:
`sudo -u pijuice python3 /usr/bin/pijuice_cli.py`

virtualarchitectures, 2022-02-07:
Thanks, I'm running RaspiOS 64-bit so that resolved the issue for me.

closed 2022-02-07 by virtualarchitectures