PiSupply/PiJuice #820: Ref #814

clarke-baxter, 2022-02-07:
Apologies for this, but I have realised I did not ask the question properly in #814.
Basically I am looking for a laptop-style power arrangement.
When the power switch is pressed the Pi and the PiJuice come alive.
When there is no external power the Pi runs on the battery.
When the power is plugged in the Pi needs to continue uninterrupted and the battery needs to charge.
When the power supply is unplugged the Pi continues on battery power.
When the power switch is pressed the Pi and the PiJuice power down.

Is this possible?

Thank you.

tvoverbeek, 2022-02-07:
I thought I answered this already, but i seems I did not.
Connect your external push button to J5 of the HAT which is in parallel to SW1
(https://github.com/PiSupply/PiJuice/tree/master/Hardware#buttons)
Program SW1 as in the defaults: SINGLE PRESS - hard power on, LONG PRESS1 SYS_FUNC_HALT_POW_OFF.
You night want to leave LONG PRESS2 (hard power off) as a last resort power off.
The parameter values are in msec.
External power should then be connected to the PiJuice micro-usb.
This should do what you want.
The power off after LONG PRESS1 is not immediate. The service will initiate an orderly shutdown and the PiJuice will remove the 5V to the Pi after 30 sec (sorry, valye hard coded in the firmware).
When no external power is present the PiJuice will then go to low power mode.
It only shuts down completely when there is no external power and no battery power.

clarke-baxter, 2022-02-07:
I'm sorry to have bothered you again. I have yet to have hands on with the PiJuice. I wasn't sure I made my question clear or fully understood the answer.

I won't bother you again.

closed 2022-02-07 by clarke-baxter