PiSupply/PiJuice #822: PiJuice Settings: "Failed to use I2C Bus. Check if I2C is enabled"

poseidonphotos, 2022-02-08:
Hi,

I have mounted and am launching PiJuice for the first time on my raspberry pi 4 (on 64 bits OS). 

After successfully installing PiJuice gui (sudo apt-get install pijuice-gui), the results are not as expected as explained here (https://learn.pi-supply.com/make/pijuice-quick-start-guide-faq/#software-installation). 

Issues:
1. New icon in the system tray does not get displayed
2. When selecting PiJuice settings from Menu>Preferences, I get an error: Failed to use I2C bus. Check if I2C is enabled

I have run the following command with results:
poseidon@raspberrypi:~ $ python
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import PiJuice
>>> pijuice=PiJuice(1, 0x14)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/pijuice.py", line 1570, in __init__
    self.interface = PiJuiceInterface(bus, address)
  File "/usr/lib/python3/dist-packages/pijuice.py", line 23, in __init__
    self.i2cbus = SMBus(bus)
PermissionError: [Errno 13] Permission denied
>>> 

Could you kindly help on what the cause of the issue is here?

Thanks

tvoverbeek, 2022-02-08:
Your default user (pi?) does not have access to the I2C bus, hence the permission error.
Make sure your default user belongs to the i2c group.
What is the output from `ls -l /dev/i2c*`?

poseidonphotos, 2022-02-08:
Ok, I tried to grant access to the I2C via:
sudo usermod -a -G i2c poseidon

After which I have run sudo usermod -a -G i2c poseidon
and the result is this:
poseidon@raspberrypi:~ $ ls -l /dev/i2c*
crw-rw---- 1 root i2c 89,  1 Feb  8 22:32 /dev/i2c-1
crw-rw---- 1 root i2c 89, 20 Feb  8 22:32 /dev/i2c-20
crw-rw---- 1 root i2c 89, 21 Feb  8 22:32 /dev/i2c-21

I still get the same original error

poseidonphotos, 2022-02-08:
Here it is:
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

poseidonphotos, 2022-02-08:
It seems I didn(t have i2c tools properly installed. After installation, I get the following output:
pi@raspberrypi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   

I still have the original errors reported.

poseidonphotos, 2022-02-08:
To add more here.

When I do pijuice_cli with the account "poseidon" and select "Status" on the screen, I get the following error:

poseidon@raspberrypi:~ $ pijuice_cli
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2074, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 287, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 385, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 790, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 827, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 416, in <lambda>
    wrapper = lambda: self.parse_input(
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 515, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 412, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 513, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 598, in keypress
    return self.top_w.keypress(self.top_w_size(size,
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1626, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2316, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 968, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 543, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 461, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1981, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 209, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 269, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 215, in get_status
    status = pijuice.status.GetStatus().get('data', {})
AttributeError: 'NoneType' object has no attribute 'status'

However, when I run the same command with account "pi" and check the status, I have no issues...

poseidonphotos, 2022-02-10:
poseidon@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --

tvoverbeek, 2022-02-10:
That looks fine. User poseidon is allowed to access the I2C bus.
Can you run the following python script:
```
from pijuice import PiJuice
pj=PiJuice(1,0x14)
print(pj.status.GetStatus())
```
Output should look like this (with PiJuice connected of course):
`{'data': {'isFault': True, 'isButton': False, 'battery': 'NORMAL', 'powerInput': 'NOT_PRESENT', 'powerInput5vIo': 'PRESENT'}, 'error': 'NO_ERROR'}`

poseidonphotos, 2022-02-11:
poseidon@raspberrypi:~ $ from pijuice import PiJuice
pj=PiJuice(1,0x14)
print(pj.status.GetStatus())
-bash: from: command not found
-bash: syntax error near unexpected token `('
-bash: syntax error near unexpected token `pj.status.GetStatus'

poseidonphotos, 2022-02-11:
@tvoverbeek , this is what I get

poseidon@raspberrypi:~ $ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import PiJuice
>>> pj=PiJuice(1,0x14)
>>> print(pj.status.GetStatus())
{'data': {'isFault': False, 'isButton': False, 'battery': 'NOT_PRESENT', 'powerInput': 'PRESENT', 'powerInput5vIo': 'NOT_PRESENT'}, 'error': 'NO_ERROR'}

tvoverbeek, 2022-02-11:
This looks fine.
If the output is correct you have no battery connected and you power the PiJuice/RPi via the PiJuice microUSB.
Correct?
So user poseidon can communicate with the PiJuice without errors.
Next step: try to run pijuice_cli manually:
`sudo -u pijuice python3 /usr/bin/pijuice_cli.py`
I do not know if under ubuntu your user poseidon is allowed to use sudo.

poseidonphotos, 2022-02-11:
PiJuice/Rpi is connected via PiJuice microUSB from a portable battery right now. 

I can run pijuice_cli manually. However, the Status shows the voltage fluctuating and the battery % remains at 0. Also, when I unplug the microUSB, everything shuts down. We should expect it to stay on, as that is the whole purpose of an UPS?

[image]

tvoverbeek, 2022-02-11:
The battery is not connected to the PiJuice.
By default No battery turn on' is disabled, hence when youremove the power the Pi goes off (no 5V to the Pi).
If this is a new PiJuice, have you removed the plastic tab between the battery connector and the battery?

poseidonphotos, 2022-02-11:
Yes, this is a new PiJuice and I have removed the plastic tab between the battery connector and the battery.

I just removed the battery and put it back in. Voila! It is working now. I guess the battery contacts were just bad.

Thanks for your help again.

closed 2022-02-11 by poseidonphotos