PiSupply/PiJuice #824: Pi Won't Remain Shutdown

oscarmenz, 2022-02-09:
I'm using a 5V USB-C connection to my Pi 4 running Ubuntu 20, and simply want the PiJuice to issue a graceful shutdown whenever the USB-C is lost, then immediately boot back up when USB-C re-connected to the PI (which it should handle on its own).
I have "System Events" > "No Power" set to "SYS_FUNC_HALT_POW_OFF". Then, in "System Task" I only have "System Task Enabled" and "Software Halt Delay" set to "60".

When I remove USB-C, the Pi successfully shuts down after about a minute. But then, 4 minutes later, it boots back up. Only to shutdown again after a minute, and repeat.

Why does it keep waking up?

tvoverbeek, 2022-02-09:
Check if you have a periodic alarm set which can wake up the Pi.
I assume this periodic rebooting happens when on battery.
Note: the software halt delay does not apply to SYS_FUNC_HALT_POW_OFF but only to a shutdown initiated by a script or from the desktop.

oscarmenz, 2022-02-09:
Under "Wakeup Alarm", "Wakeup Enabled" is not selected? 

I am now trying a different approach, where I simply trigger a python script to handle the shutdown when "No Power" event fires. However, I can't seem to get the script running.

I have set "No Power" to USER_FUNC1 and defined USER_FUNC1 as "/home/ubuntu/no_power_test.py"

The script is in /home/ubuntu/no_power_test.py and is owned by ubuntu, and I can run with ./no_power_test.py

I have checked that ubuntu is a member of i2c and pijuice user groups

tvoverbeek, 2022-02-10:
Is the file 020_pijuice-nopasswd present in /etc/sudoers.d?
The userfunc is called by the service which is running as user pijuice and tries to execute the userfunction
with 'sudo -u <user> <userfunction>`. User pijuice needs the permission to use 'sudo -u <user>'.
You can test this by stopping the service `sudo systemctl stop pijuice` and running the service manually in a terminal: `sudo -u pijuice python3 /usr/bin/pijuice_sys.py` and see if you get an error message when the no power event triggers.