PiSupply/PiJuice #826: Add tags/releases

cglewis, 2022-02-10:
Great product - really enjoying it so far. One pain point is there are no tags or releases for the project so in order to get the debian package and firmware, cloning has to include whatever was committed most recently, which means caching a git clone isn't possible if/when updates are pushed to the repo. If tags were created when new versions are made then the repo could be cloned with the `-b <tag>` option which would ensure cache is preserved even as new changes are made to the repo.

tvoverbeek, 2022-02-11:
Just added a tag (V1.8) to the current state of the main branch which corresponds to the just released 1.8 packages.
Will see what we can do about also including Releases

cglewis, 2022-02-11:
Awesome, thank you!

closed 2022-02-11 by cglewis