PiSupply/PiJuice #829: What is needed to use the PiJuice with the official touchscreen display?

K9LOL, 2022-02-13:
I’ve tried various wiring configurations to get the PiJuice to work with the official 7” touchscreen display, but with no luck. The screen will initially power on OK, then fade to black and not come back on. It appears to be a power issue, but none of the wiring configurations I’ve tried have worked. 

Here’s how I currently have it configured:

- Power in to PiJuice
- 5V to display via GPIO
- Ribbon cable connected between Pi and display 

That seems to work “best,” but still fails partway through boot (at which point the green light on the pi flickers continuously). 

I’m using the stock BP7X/1820mAh battery. 

Thank you!

K9LOL, 2022-02-14:
I should add that this is using a 5V 3A power supply.

K9LOL, 2022-02-22:
Thank you for your help. I now have it set up as follows: Nitecore battery -> Display -> RPi

I’m using the USB out on the display to go directly to the RPi, as recommended in the Display documentation. This allows everything to power on properly with no low voltage warnings, BUT the PiJuice doesn’t show any indication it is taking a charge. Settings shows:  Battery: NORMAL, GPIO: ~5V, -2.4A, NOT_PRESENT. 
HAT Settings is Inputs precedence: 5V_GPIO, GPIO Input Enabled is on, Power regulator mode: POWER_SOURCE_DETECTION. 

Do I have a setting incorrect, or is there something else I need to check? Thank you.

K9LOL, 2022-02-22:
That improves the situation but occasionally on boot the system still hangs with a flickering green light on the RPi. Once it’s in this state I have to remove power from the main board _and_ then use SW2 to do a hard func power off. I guess it’s still drawing more than the power supply can handle.

My hope for this setup was eventually to get the whole thing powered off a battery (a 10000 mAh Nitecore power bank), but if I have the PiJuice in the mix I can’t even seem to get it to work reliably on AC. If I take the PiJuice out of the mix then everything seems to work fine, even off the Nitecore battery—but of course I lose the benefits of the PiJuice. 

Are there any recommendations in terms of power configuration to get this working reliably?

Stevo52, 2022-04-16:
I have the same configuartion of RPi 4B  (Raspbian v11 Bullseye) and PiJuice HAT with either the standard BP7X or a larger external battery) and the RPi 7" Touchscreen which is powered of the GPIO pins (from the P3 pins)  and I also have a couple of RelayPlate boards and an Adafruit GPIO Expander connected. It runs fine off the standard Rasperry power supply connected to the PiJuice. The PiJuice has a button connected at SW1 pins and the standard PiJuice button functions work as they should (but I do shut down the RPi from the GUI, then power off via the button long press.

If the green led on the RPi is still flickering after the system has booted to idle, then something is running in the background?