PiSupply/PiJuice #833: High current reported in DCDC mode

nmrkr, 2022-02-17:
In POWER_SOURCE_DETECTION and LDO mode the CLI reports an idle current of around 650mA, which is reasonable, and wobbles around a bit over time. In DCDC mode however the current display jumps to around 2.4A and stays at one particular numerical value. No wobble at all. The value also stays constant under heavy load except every now and then it will jump to one other value around 800mA.

Did I miss something in the documentation? Does current measurement simple not work in DCDC mode? Or would this indicate a hardware defect?

The hardware configuration is Pi4, PiJuice and stock BP7X; no additional peripherals connected. Firmware version 1.6.

mmilann, 2022-02-17:
@nmrkr 5V GPIO current measurement is only possible in POWER_SOURCE_DETECTION detection or LDO mode. For Pijuice version V2.2 current measurement is calibrated around 50mA (to be used for estimating when Rpi is off to enter standby mode) and is rough estimation above 50mA. LDO mode is most precise for current measurement but  least efficient around 5% power dissipation compared to DCDC or POWER_SOURCE_DETECTION modes. 

New PiJuice revision V2.3.0 introduce precise current measurement to mAmps.

nmrkr, 2022-02-22:
Thank you for cleaing that up! I think this should be documented (unless I missed it) and ideally the CLI should not report random currents when the actual value simply cannot be known. Oh well, nitpicking :smiley: