PiSupply/PiJuice #834: shutdown bug with system tasks (pijuice self restarts)

OhSoGood, 2022-02-20:
Hi,

I intermittently face the same issue on all my platforms (I have 40+, see details below): when I switch power off for the pijuice, either by script (with a `pijuice.power.SetPowerOff(20)` followed by a `os.system("sudo halt")` ), or by long pressing 20 seconds the SW1 button, the power gets off as expected and then comes back on a few seconds after. This is only happening when both 1. systems tasks are enabled, 2. USB power cable is connected to the pijuice and 3. I have a second hat on the raspberry (a Waveshare 7600EX 4G modem).

Once it happens on a platform, it will continue to happen forever: I cannot turn it off anymore. To kind of "solve" it, I can either remove the USB power cable from the pijuice (not a real solution...), remove the 4G hat (not a solution also!). Or a third solution is to uncheck everything under System Task, to reboot the pijuice (I ignore if that's necessary), and to set back System Tasks options.

The bug like the hacks to fix it do not seem very logical, but that's the best I found so far.

Can anybody help? Thank you!

My platform runs on Raspberry Pi Lite of January 2022 (with debian Buster 10.11), with either pijuice 1.7 (and firmware 1.5) or pijuice 1.8 (and firmware 1.6). The Waveshare 7600EX 4G modem required "dtoverlay=i2c-rtc,ds1307" in /boot/config.txt (confirmed by a "UU" with "i2cdetect -y 1"). The pijuice config is:
- No battery turn on: checked
- System Task enabled: checked
- Wakeup on charge: checked. Trigger level: 15, Restore: checked
- Min charge: checked. Threshold: 6

khufkens, 2022-02-20:
Had similar issues and setting:
```
pj.power.SetWakeUpOnCharge(100)
```
kind of solved it a hackish way.

tvoverbeek, 2022-02-20:
As @khufkens writes your problem is the wakeup-on -charge setting:
_Wakeup on charge: checked. Trigger level: 15, Restore: checked_
With USB power to the PiJuice it detects charging and starts the Pi again.
Thus Pi restarting will occur as soon as the battery charge level is above 15%.
Disable the wakeup-on-charge setting (both level and restore).
Why do you set it in the first place if you want the Pi to stay off?

khufkens, 2022-02-20:
Because there doesn't seem to be python API call to deal with disabling it. I need control over the state of the Pi(Juice) - i.e. when a user manually alters something a script should overwrite these settings by default to create a reproducible baseline.

khufkens, 2022-02-20:
Didn't see the 'DISABLED' statement... my bad.

OhSoGood, 2022-02-20:
Hi,

Thank you for your answers but I honestly admit that I don't understand.

With always the same configuration, including _Wakeup on charge: checked. Trigger level: 15, Restore: checked_, and with the battery always charged at 95-100% plus external power plugged to the USB entry of the Pijuice, I can actually fully power off the pijuice most of the time. It's only on some seldom cases that it starts again. Whatever the actual configuration the settings lead to, it should be constant and always the same, shouldn't it?