PiSupply/PiJuice #835: SYS_FUNC_HALT_POW_OFF vs SYS_FUNC_SYS_OFF_HALT

liparrok, 2022-02-21:
Hey guys,

i wonder what is difference between SYS_FUNC_HALT_POW_OFF and SYS_FUNC_SYS_OFF_HALT?

and second thing, I just updated pijuice from V1.3 to V1.6 and I see now there is couple new options under System Events, that are not described here: https://github.com/PiSupply/PiJuice/tree/master/Software. When can I expect that?

Thanks for your answers, Rok

tvoverbeek, 2022-02-21:
SYS_FUNC_HALT_POW_OFF first switches off VSYS (available on the P8 conenctor), schedules the 5V power off to the Pi via the GPIO after 60 seconds and then starts the shutdown.
SYS_FUNC_SYS_OFF_HALT switches off VSYS and then starts the shutdown. The 5V to the Pi stays on.
Note, both functions are executed by the pijuice service running on the Pi.

The new events are 2 hooks to the start and finish of the service which you could use for your own startup.shutdown actions. Note the sys_stop event is executed when the shutdown is already in progress.

I know the software README still needs updating, also for the system task (watchdog and wakeup-on-charge settings can be stored in non-volatile memory)

liparrok, 2022-02-22:
Correct me if I am wrong please, so VSYS is output from BUCK converter (battery charger), right (P3 in think, 6pin connector)?

Many thanks for your answers. 
Rok

tvoverbeek, 2022-02-22:
Yes, correct and yes the VSYS available on P3.

liparrok, 2022-02-22:
Thanks again!

closed 2022-02-22 by liparrok