PiSupply/PiJuice #836: Troubleshooting Auto-shutdown within PiJuice HAT Config

poseidonphotos, 2022-02-21:
I have setup auto-shutdown of my raspberry pi by setting up a system task/system event in my PiJuice HAT config.

System task: Set minimum charge to 96%
System Events: Set low charge to USER_FUNC1
User scripts: Set file path (/home/pi/myshutdown.py)

whereby the python script it below.

I have disconnected the power supply and waited for the power to dip down to 95% (i.e. below 96%); however, the raspberry pi does not shut down. 

Why is that?

----
#!/usr/bin/python3

import os
import logging
from pijuice import PiJuice

logging.basicConfig(
	filename = '/home/pi/pistatus.log',
	level = logging.DEBUG,
	format = '%(asctime)s %(message)s',
	datefmt = '%d/%m/%Y %H:%M:%S')

pj = PiJuice(1,0x14)

# Set wakeup_enabled and wakeup_on_charge just to be sure
pj.rtcAlarm.SetWakeupEnabled(True)

# The Raspberry Pi should wake up even when there is no battery, i.e. a
# battery percentage of 0
pj.power.SetWakeUpOnCharge(0)

# Write statement to log
logging.info('Raspberry Pi battery running low')

# Show custom script is run by blinking the user LED red 10x
pj.status.SetLedBlink('D2', 10, [200,0,0], 50, [0, 0, 0], 50)

# Now shut down
os.system("sudo shutdown -h now")

tvoverbeek, 2022-02-21:
Do you get anything in your log file?
What happens if you run your user script manually?
Is the myshutdown.py script executable?

poseidonphotos, 2022-02-27:
Thanks for this.

Log is created as expected:
27/02/2022 23:34:21 Raspberry Pi battery running low

And the script does run as expected when run manually...(which means the python script is executable).

Even after the log is created, the pi  keeps on running. It seems the "shutdown" command is never executed...

tvoverbeek, 2022-02-27:
How do you run the script manually?
It should run with just `/home/pi/<yourscript>` (assuming it is located in /home/pi), not with `python3 /home/pi/<yoursrcipt>`.

poseidonphotos, 2022-02-28:
Ok, I ran it like you have suggested and this is what I got:

pi@raspberrypi:~ $ /home/pi/myshutdown.py
Connection to 192.168.11.29 closed by remote host.
Connection to 192.168.11.29 closed.

It seems to have shut it down, but via the pijuice configuration, the python script doesn't seem to get kicked off?

poseidonphotos, 2022-03-02:
@tvoverbeek 

I am retesting things. My findings:
1. /home/pi/myshutdown.py is executable. I can manually shut the pi down via this script
2. However, when the "low charge" threshold set in pijuice config is hit, I am not seeing anything in the log (sorry, I was wrong earlier). This would mean that my pijuice config settings is wrong? It is never running the python script

I have followed all instructions mentioned here (https://raspberrypi-guide.github.io/other/boot-automation-pijuice#turning-the-raspberry-pi-on-and-off-automatically) under "Turn off Raspberry Pi when battery too low"

tvoverbeek, 2022-03-02:
I just copied the myshutdown.py script from https://raspberrypi-guide.github.io/other/boot-automation-pijuice#turning-the-raspberry-pi-on-and-off-automatically under "Turn off Raspberry Pi when battery too low". and set the low charge level a few % below the current level. Removed the external power, so the Pi was running on battery only. Pi shut down at the specified charge level with entry in the log file.
I used the following settings
On System Task panel: 'System Task enabled' checked, "Min charge' selected with threshold 95, 'Software Halt Power Off' selected with Delay period 20 seconds
On System Events panel: 'Low charge:' selected with parameter USER_FUNC2.
On User Scripts panel: 'USER FUNC2' defined as '/home/pi/myshutdown.py

This is the way to connect your script to the low charge event (not detailed on the raspberrypi-guide site).
One way to test is to comment out the last line (the shutdown line), then the script should run each time the system task (= the pijuice service) checks the charge level when it is below the threshold (= every 5 seconds) and add a line to the log file.

poseidonphotos, 2022-03-03:
Thanks. What you have mentioned is exactly how I have it setup, but for some reason the python script never gets triggered, when the battery % dips below the minimum charge threshold. Is there some other setting I am missing here?

[image]

[image]
[image]
[image]

tvoverbeek, 2022-03-03:
Then the only (?) reason it does not work is tha pijuice service is not running.
Check the output of `systemctl status pijuice`.
It should look something like this:
```
pi@rpi32-deb10:~ $ systemctl status pijuice
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2022-03-03 14:10:00 CET; 1min 25s ago      <====
 Main PID: 400 (pijuice_sys.py)
    Tasks: 1 (limit: 3720)
   CGroup: /system.slice/pijuice.service
           └─400 /usr/bin/python3 /usr/bin/pijuice_sys.py

Mar 03 14:10:00 rpi32-deb10 systemd[1]: Started PiJuice status service.
Mar 03 14:10:06 rpi32-deb10 pijuice_sys.py[400]: RTC os-support OK
pi@rpi32-deb10:~ $ 
```
Alternatively check the output of `ps ax | grep pijuice_sys | grep -v grep`:
```
pi@rpi32-deb10:~ $ ps ax | grep pijuice_sys | grep -v grep
  400 ?        Ss     0:00 /usr/bin/python3 /usr/bin/pijuice_sys.py
pi@rpi32-deb10:~ $ 
```

poseidonphotos, 2022-03-03:
Something is blocking the script?

pi@raspberrypi:~ $ systemctl status pijuice
● pijuice.service - PiJuice status service
     Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-03-02 22:01:49 JST; 24h ago
   Main PID: 469 (pijuice_sys.py)
      Tasks: 1 (limit: 8987)
        CPU: 3min 17.461s
     CGroup: /system.slice/pijuice.service
             └─469 /usr/bin/python3 /usr/bin/pijuice_sys.py

Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed
Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed
Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed
Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed
Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed
Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed
Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed
Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed
Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed
Mar 03 21:34:58 raspberrypi pijuice_sys.py[469]: root owned /home/pi/myshutdown.py not allowed

tvoverbeek, 2022-03-03:
That is it: **User scripts shall not be owned by root for security reasons**.
It should be owned by a non-privileged user belonging to the pijuice group.
The default 'pi' user already belongs to the pijuice group.

poseidonphotos, 2022-03-03:
Ok got it. Thanks for your patience with this! Works now :)

closed 2022-03-03 by poseidonphotos