PiSupply/PiJuice #837: Raspberry keep powered after shutdown

gremlin-it, 2022-02-21:
Hi all, 

I have a small but annoying issue, probably depend on some mis-configuration, but don't know what can be. 

It's quite simple, after "shutdown" command issued by root on a terminal, power is kept forever (i wait for more than an hour). 
I expect the pijuice to cut the power after few minutes after the raspberry finish the shutdown. 

I'm using PiJuice Zero (fw 1.6) on a Raspberry Pi 3B+ 
powered via J4 with a 9V power plug and two 18650 LiPo battery in parallel on J2. 

with the pijuice_gui tool I setup the battery profile and : 

System Task: 
(checked) System Task Enable 
(checked) Software Halt Power Off 30 seconds 

System Events, none checked

under Configure Hat: 
Buttons: 
SW1: LONG_PRESS: SYS_FUNC_HALT_POWER_OFF: 3000ms 

When SW1 is pressed for 3 secods, raspberry shutdown correctly and power if cut about 30 seconds after the button was pressed, as expected. 

Whit shutdown issued power is kept forever. 

thanks for help

piPixel29, 2022-02-22:
Same concern here !

tvoverbeek, 2022-02-22:
Which OS? Works for me using your settings on Raspberry Pi OS bith buster and bullseye.

A380Coding, 2022-04-01:
I have the same concern. Providing a shutdown command, pijuice stays active for ever. I enabled System Events -> No Power with SYS_FUNC_HALT_POW_OFF as a work around since power is disconnected automatically based on load in my case. Worked the first time, but testing again did not work again. Raspberry remains on battery and power is not cut. Any ideas?

tvoverbeek, 2022-04-01:
The SYS_FUNC_HALT_POW_OFF waits 30 seconds before renmoving the power.
If during this wait external power returns, the 30 second countdown is cancelled and the PiJuice will continue to deliver 5V to the GPIO.
Could this be your case?

A380Coding, 2022-04-01:
I played around with it. I had the power connected the Pi in order to boot it up after power is reestablished. I think that caused the issue. Connected to the PiJuice USB port it is working. Initially not booting after power is connected but I solved that with the  Restore option in the System Task settings.
For me it is working now as expected, thanks for the reply!

But I found an issue in the commandline tool. When "Restore" is active for "Wake on charge" then changing the "Trigger level" value is not working. It will always show the old value. Once "Restore" is deactivated it magically shows the new value. Value is being used, but just not shown when "Restore" is active.

tvoverbeek, 2022-04-01:
@A380Coding Could you please raise a separate issue for the "Wakeup on charge" not changing when "Restore" is active?
Just use your last paragraph.
Makes it easier to trace the issue.
Then we can close this issue.
Thanks in advance.

tvoverbeek, 2022-04-01:
New issue raised (#852).
Closing this one

closed 2022-04-01 by tvoverbeek