PiSupply/PiJuice #838: PiJuice Fails Upon Copying To Smartphone Via USB-A

Al-Ascough, 2022-02-21:
I have a simple Raspberry Pi 4 application that copies .jpg files to a smartphone attached via USB-A. This application runs successfully when powered by wall outlet.

However, when attempting to run the application while powered by PiJuice, the Raspberry Pi 4 appears to shut off. Additionally, the RaspberryPi also appears to shut off after roughly eight minutes of simply idling with the smartphone plugged in, but does not behave this way without the smartphone. Finally, the system fails to restart, despite having charge well in excess of the chosen "Wake up on charge" value, i.e. 20%.

I suspect this has something to do with the smartphone's power draw. I will attempt a data-only USB connection later today, but do not expect it to work as others inform me a powered connection is necessary for data sharing on smartphones. Therefore, I anticipate that I'll need a solution based on PiJuice's settings.

tvoverbeek, 2022-02-21:
I  am afraid the PiJuice cannot deliver sufficient current to both the Pi and the USB-A connected smart phone.
Max current on 5V GPIO delivered by the PiJuice is 2.5A provided the battery has sufficient capacity.
The standard BP7X does not have sufficient capacity for both Pi4 and phone.

Al-Ascough, 2022-02-21:
> I am afraid the PiJuice cannot deliver sufficient current to both the Pi and the USB-A connected smart phone. Max current on 5V GPIO delivered by the PiJuice is 2.5A provided the battery has sufficient capacity. The standard BP7X does not have sufficient capacity for both Pi4 and phone.

Please forgive the noob question, but could this be resolved with a bigger battery? Or is the current limited by the HAT itself?

tvoverbeek, 2022-02-21:
The 2.5A limit is due to the power converter on the board. So a bigger battery will not help.
There are plans for an update such that the PiJuice could deliver 3.5A (same as the recommended power supply for the Pi4), but I have no idea when or if this will actually materialize. You will have to contact Pi-Supply for this.

Al-Ascough, 2022-02-21:
> The 2.5A limit is due to the power converter on the board. So a bigger battery will not help. There are plans for an update such that the PiJuice could deliver 3.5A (same as the recommended power [...]

Thank you. You've probably saved me a lot of time. Is there an upvote button? (I'm new to Github. Are the rules anything like Stack Overflow?)