PiSupply/PiJuice #840: Add a "poweroff" action to the icon menu

piPixel29, 2022-02-22:
Hi
Would it be possible to add a "clean poweroff" entry in the pijuice icon contextual menu on the desktop ?
This would trigger an immediate SYS_FUNC_HALT_POW_OFF action.

I'm looking for a clean way to turn off the Pi.
My Pi is powered from USB-C. The PiJuice gets its power from GPIO. The PiJuice settings are configured to trigger a SYS_FUNC_HALT_POW_OFF when battery reaches <40%.
If I turn the Pi off via a usual "poweroff" or "shutdown", it will turn off but the PiJuice will keep it powered. My Pi has a USB device connected wich I need to be unpowered as soon as the pi is off. The PiJuice keeps this USB device alive as long as the battery can.
Eventually the battery reaches 0% and dies, which is bad for battery and erases the clock ...

Or maybe the pijiuce settings could allow us to set a SYS_FUNC_HALT_POW_OFF upon reception of a configurable kill signal ??

Thanks for reading

piPixel29, 2022-02-22:
Hi, Thanks for your response.

I was not clear enough.
The Pi is powered from USB-C, I shut it down using usual linux poweroff, then unplug the USB-C.
In this scenario the battery continues to deliver power to the pi, until it drains to 0%.

So I'd like to be able to unplug the USB-C alim, then click the PiJuice applet menu "clean poweroff" and let the SYSFUNCHALTPWOFF hapen.

My only solution yet is to remove the USB-C alim and let the battery drain until it reaches the lowbatt threshold which triggers the SYS FUNC HALT PWOFF. Which is not cool because it makes me set a "high" threshold like 60% and thus makes the pi turn down too fast in case of a normal power outage.

tvoverbeek, 2022-02-22:
You can do the following:
In the system task panel enable 'Software Halt Power Off' with an appropriate time out.
(SYS_FUNC_HALT_POW_OFF uses 30 sec).
Then after removing the UCB-C power do a shutdown either from a terminal or the desktop.
The 5V from the PiJuice to the Pi will go off after the specified timeout.
Note: if the PiJuice sees the external power returning before the timeout it will cancel the timer
      and the 5V from the PiJuice will stay on, hence remove the USB-C before starting the shutdown.