PiSupply/PiJuice #842: Issue Running Python Script on Restart

poseidonphotos, 2022-03-04:
I want my PiJuice to automatically run a python script when the battery level hits a certain threshold. I am doing the following:

1. In System Task: Set system task enabled and set wakeup on charge to have a trigger level, e.g. 90%
2. In System Events: Set System start to USER_FUNC1
3. Under User scripts: Set User FUNC1: /home/pi/moneroOnRestart.py

My python script includes a line as per below:
os.system("sudo setsid ./xmrig -o gulf.moneroocean.stream:10128 -u blahblah -p miningMonero1")

However, when I manually run the python script, I get an error:
setsid: failed to execute ./xmrig: Permission denied

But the command itself can be run in the terminal with no issues.

What could be the root cause of this?

Also, assuming that I get the script fixed, am I correct to say that the script should run when the battery level hits the trigger level?

Thanks

poseidonphotos, 2022-03-05:
I understand this is a permission issue but I think I have the right permissions?

pi@raspberrypi:~ $ getfacl xmrig
# file: xmrig
# owner: pi
# group: pi
user::rwx
group::r-x
other::r-x

pi@raspberrypi:~ $ getfacl moneroOnRestart.py 
# file: moneroOnRestart.py
# owner: pi
# group: pi
user::rwx
group::r-x
other::r-x

tvoverbeek, 2022-03-05:
For an explanation see https://stackoverflow.com/questions/25701333/os-setsid-operation-not-permitted.
Why are you using setsid? Use the subprocess module instead

poseidonphotos, 2022-03-05:
Ok, that works. Thanks for your help everytime :)

closed 2022-03-05 by poseidonphotos