PiSupply/PiJuice #844: Pijuice without any GUI | Voidlinux

numerys, 2022-03-11:
I installed [Voidlinux](https://voidlinux.org) onto a CM4. I tried to get the RTC working and added the line

```dtoverlay=i2c-rtc,ds1339,addr=0x68,trickle-resistor-ohms=4000```

to

```/boot/config.txt```

After that I finally got a

```
[    4.246880] rtc-ds1307 1-0068: registered as rtc0
[    4.248100] rtc-ds1307 1-0068: setting system clock to 2000-01-04T11:56:05 UTC (946986965)
```

And only once I could get a successful

```hwclock -r```

Mostly I got

```
hwclock -r                                            
[ 4547.668586] rtc-ds1307 1-0068: read error -121
hwclock: ioctl(3, RTC_UIE_ON, 0) to /dev/rtc0 failed: Remote I/O error
```

I tried different values like ```2000``` or ```250```. Same result.

So I abandoned the idea to get the Pijuice running without Pijuice-software.

I installed python3, python3-smbus too. I copied ```pijuice.py``` and ```pijuice_util.py``` into a directory and tried/got

```
# python3 pijuice_util.py --get-config
ChargingConfig:  COMMUNICATION_ERROR
Battery Profile:  COMMUNICATION_ERROR
Firmware Version:  COMMUNICATION_ERROR
```

What else could I do to get the Pijuice working with CLI and Voidlinux?

Further Info:

```
# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

# i2cdetect -y 10
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- 0c -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 2f 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- 51 -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

numerys, 2022-03-11:
I forgot to mention that I use a CM4 with RPi IO board.

numerys, 2022-03-11:
Added ```i2c_bcm2835.debug=3``` and with ```hwclock -r``` gives
```
[  198.819596] bcm2835_i2c_xfer: msg(1/2) write addr=0x68, len=1 flags= [i2c1]
[  198.819619] bcm2835_i2c_xfer: msg(2/2) read addr=0x68, len=7 flags= [i2c1]
[  198.820630] start_transfer: msg(1/2) write addr=0x68, len=1 flags= [i2c1]ff]
[  198.820645] isr: remain=1, status=0x10000055 : TA TXW TXD TXE  [i2c1]
[  198.820659] start_transfer: msg(2/2) read addr=0x68, len=7 flags= [i2c1]
[  198.820673] isr: remain=7, status=0x132 : DONE TXD RXD ERR  [i2c1]          
[  198.820691] i2c-bcm2835 fe804000.i2c: i2c transfer failed: 100
[  198.826681] rtc-ds1307 1-0068: read error -121
```

tvoverbeek, 2022-03-12:
i2cdetect output looks OK.
Bus 0 address 0x50 is the PiJuice HAT EEPROM
Bus 1 0x14 is the PiJuice mcu, 0x68 is the simulated ds1339 RTC in the mcu.
What are te 3 units on bus 10? Is this a software i2c? Or a real one on the CM4?
Could transactions on bus 10 interfere with transactions on bus 1?
I do not know Voidlinux, but does it have the same hardware drivers as Raspberry Pi OS?
Have you tried with Raspberry Pi OS on your CM4?

closed 2022-03-15 by numerys