PiSupply/PiJuice #845: Low corrent using J4 contacts

tomasz90, 2022-03-12:
I am experiencing issue when I am using J4 for powering hat. I have around 1,1 A max current - which is not limited by power supply checked with two psus. When using microusb I am getting 2,5-2,6A which is correct - my battery has large capacity around 10000mAh - so I limited charging speed to 2000mA. Is there any solution for this problem? Any settings that may fix this?

tvoverbeek, 2022-03-12:
I do not understand. If you look at the schematic (https://github.com/PiSupply/PiJuice/blob/master/Documentation/PiJuice_Schematic.PDF) you see that the J4 pins are in parallel to the microUSB power connections. So the results should be identical.
The only difference could be the difference (e.g. resistance) in the power connection cables.
How do you measure the current?

tomasz90, 2022-03-13:
I measured the current via usb stick. Yep after some investigation I figured out that usb power lines are the same as in J4. 
I also tried different USB cable, I found the cheap one without D+ and D- lines, then the current I measured was around 0,8A. Maybe current draw is negotiable via data lines?

tvoverbeek, 2022-03-13:
Note that on the schematic the D+ and D- lines are connected to zener diodes.
The current difference you are seeing is caused by (useless) current going through the zener diodes.