PiSupply/PiJuice #846: pijuice_util.py - hot to configure the Pijuice from commandline

numerys, 2022-03-15:
I've set up pijuice_util.py. It seems to work but Idk how to _set_ the ```wakeupcharge```?

A next question are about the user_funcs. What is e.g. ```USER_FUNC1```? How to read that string the moment such event occurs?

I'm accessing the RPi via serial/commandline. I have no GUI available.

I got some outputs similar like. 

```
# python3 pijuice_util.py --get-status

{ 'data':  { 'isFault':        True
           , 'isButton':       True
           , 'battery':        'CHARGING_FROM_IN'
           , 'powerInput':     'PRESENT'
           , 'powerInput5vIo': 'NOT_PRESENT' }
, 'error': 'NO_ERROR' }

# python3 pijuice_util.py --get-config

ChargingConfig:   {'charging_enabled': True}

Battery Profile:  { 'capacity':           1000
                  , 'chargeCurrent':      625
                  , 'terminationCurrent': 50
                  , 'regulationVoltage':  4180
                  , 'cutoffVoltage':      3000
                  , 'tempCold':           0
                  , 'tempCool':           2
                  , 'tempWarm':           49
                  , 'tempHot':            65
                  , 'ntcB':               3450
                  , 'ntcResistance':      10000 }
                  
Firmware Version: { 'version': '1.5'
                  , 'variant': '0' }

# python3 pijuice_util.py --get-battery

{ 'batteryCurrent': -457
, 'batteryVoltage': 4165
, 'chargeLevel':    99
, 'batteryStatus':  'CHARGING_FROM_IN' }

# python3 pijuice_util.py --get-input

{ 'ioVoltage':       5039
, 'ioCurrent':       690
, 'gpioPowerStatus': 'NOT_PRESENT'
, 'usbPowerInput':   'PRESENT' }

# python3 pijuice_util.py --dump

{ "general": { "run_pin":                  0
             , "i2c_addr":                 "14"
             , "i2c_addr_rtc":             "68"
             , "eeprom_addr":              0
             , "eeprom_write_unprotected": false
             , "precedence":               1
             , "gpio_in_enabled":          true
             , "usb_micro_current_limit":  1
             , "usb_micro_dpm":            0
             , "no_battery_turn_on":       false
             , "power_reg_mode":           0
             , "charging_enabled":         true }
, "led": [ { "function": "CHARGE_STATUS"
           , "color":    [60, 60, 100] }
         , { "function": "USER_LED"
           , "color":    [0, 0, 0] } ]
, "button": { "SW1": { "PRESS":        {"function": "NO_FUNC",             "parameter": 0}
                     , "RELEASE":      {"function": "NO_FUNC",             "parameter": 0}
                     , "SINGLE_PRESS": {"function": "HARD_FUNC_POWER_ON",  "parameter": 800}
                     , "DOUBLE_PRESS": {"function": "NO_FUNC",             "parameter": 0}
                     , "LONG_PRESS1":  {"function": "SYS_FUNC_HALT",       "parameter": 10000}
                     , "LONG_PRESS2":  {"function": "HARD_FUNC_POWER_OFF", "parameter": 20000}}
            , "SW2": { "PRESS":        {"function": "NO_FUNC",             "parameter": 0}
                     , "RELEASE":      {"function": "NO_FUNC",             "parameter": 0}
                     , "SINGLE_PRESS": {"function": "USER_FUNC1",          "parameter": 400}
                     , "DOUBLE_PRESS": {"function": "USER_FUNC2",          "parameter": 600}
                     , "LONG_PRESS1":  {"function": "NO_FUNC",             "parameter": 0}
                     , "LONG_PRESS2":  {"function": "NO_FUNC",             "parameter": 0}}
            , "SW3": { "PRESS":        {"function": "USER_FUNC3",          "parameter": 0}
                     , "RELEASE":      {"function": "USER_FUNC4",          "parameter": 0}
                     , "SINGLE_PRESS": {"function": "NO_FUNC",             "parameter": 0}
                     , "DOUBLE_PRESS": {"function": "NO_FUNC",             "parameter": 0}
                     , "LONG_PRESS1":  {"function": "NO_FUNC",             "parameter": 0}
                     , "LONG_PRESS2":  {"function": "NO_FUNC",             "parameter": 0} } } }
```

numerys, 2022-03-15:
I've found (a little late) pijuice_cli.py. Nevertheless I can't find some code how to write a little daemon.py. I would like to have a daemon like

- if wakeup on charge is true
  - get trigger level
  - set trigger level if needed
- else
  - set wakeup on charge setting
  - get trigger level
  - set trigger level if needed
- while true
  - if power state is false
    - sleep 10 min
    - if power state is false
      - shutdown now 
  - else
    - sleep 10 sec

tvoverbeek, 2022-03-16:
The daemon is already there. It is the pijuice service. Look at pijuice_sys.py (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py).
THis is started as a system service by systemd.
The service definition is here: https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/pijuice.service

closed 2022-04-14 by numerys