PiSupply/PiJuice #847: Stacking Sixfab On PiJuice

Al-Ascgh, 2022-03-18:
Hi, team!

I'm attempting to stack a Sixfab hat (an LTE modem) onto PiJuice. By itself, the Sixfab is fully functional and tests just fine. 

By contrast, when staked on PiJuice, the Sixfab seems to receive power (LEDs light up) but otherwise loses its cellular connectivity. My reading so far leads me to believe the Sixfab's pinout (linked below) must somehow overlap the PiJuice's. However, I'm still quite new to Raspberry Pi, and apparently stacking hats isn't considered best practice, so I haven't found a lot of supporting documentation.

Hoping for some guidance on which pins may be interfering and if it's possible to reassign them?

Thank you for your wisdom!

Sixfab pinout: https://docs.sixfab.com/docs/raspberry-pi-4g-lte-cellular-modem-kit-technical-details

tvoverbeek, 2022-03-18:
From the pinouts thers is no concflict between the Sixfab and Pijuice.
Pijuice only uses I2C and the power pins.
If you are trying to power both the RPi and the modem from only the PiJuice your total power consumption is probably too much for the PiJuice, especially when your RPi is a Pi4.

Al-Ascgh, 2022-03-18:
That makes sense. However, inserting the PiJuice seems to interrupt the
Sixfab even while receiving supplemental power from the wall. Shouldn't
that fix any draw issues?

parietinus, 2022-04-18:
I have a Raspberry pi with Sixfab HAT and fed by Pijuice running since 2020.
Initially I had the same symptoms: sixfab started, went up on the net and then connection went down again.
Started the cycle soon thereafter once again and down again.

I resolved the isuue by: configuring the sixfab to use the QMI interface and couple to the raspberry via USB cable. Not connected via GPIO; make sure tha USB cable is high-speed quality; otherwise the problem remains.

After this: Sixfab worked reliably.
But then the system sometimes frooze after startup:  because I have also 2 camera's, and a usb-stick in the USB ports of the raspberry, which was too much it se
[image]
emed (Sixfab 1A, cams 400mA; Pi 900mA).
This problem was avoided by changing the power regulatormode on the Pijuice to DCDC.
From then on, everything worked perfectly, and it still does.
[image]

Hope this helps,
Best regards

tvoverbeek, 2022-04-18:
@parietinus Many thanks for the feedback.