PiSupply/PiJuice #853: Battery gets empty after shutdown

A380Coding, 2022-04-03:
Following scenario:
Pijuicesetting:
- Wakeup on charge active, trigger level 80%, Restore active
- Min charge active, Threshold 50%
- Software halt power delay period: 60 sec active
-System Event Low charge: SYS_FUNC_HALT_POW_OFF
-USB power connected to pi juice

Raspberry pi is on and running, >95% battery charge and power is connected and available. Raspberry pi gets a shutdown via cronjob and shuts down. After 60 seconds power to raspberry pi is off (red light on raspberry goes off). 10 minutes after that the USB power gets disconnected automatically (on purpose based on power consumption).

So far so good.
When the AC power is connected next time, usb power to the pi juice is available and I would expect the pijuice to start the raspberry pi again. This does not happen, as the battery is down to 50%. The battery now starts charging and once hitting 80% starts the raspberry pi.
Why is the battery down to 50% after short time (few hours)? Raspberry pi is shutdown. I have usb connected to the pijuice which is without power at that time and hdmi is connected.
I would expect the battery to be at >90% and starting the raspberry pi immediately after the usb power is available again. It makes sense that the battery is first charged to 80% and then starts the raspberry, but why is the battery down to 50% in the first place. What is consuming the power?

tvoverbeek, 2022-04-04:
2 possible reasons:
- Some peripheral is still sipping power when the RPi is off
-  Your battery is going bad. (My first BP7X went bad and had to be replaced)

A380Coding, 2022-04-04:
It's only the Pi with PiJuice. The USB Powresupply connected to the Pijuice USB and an HDMI cable to the Pi.
Not sure how that should happen. Additionally it stops at 50% which I set as a threshold. So I do not think, that the battery is bad. And the only way to sip battery would be via the HDMI or the USB power supply. But I would not understand why it then would stop at 50%. Does not make any sense.

tvoverbeek, 2022-04-05:
What action/function have you specified for 'low charge'?
It seems at 'low charge' the 5V to the Pi is actually removed.
Also after the cron based shutdown what is the charge status led on the PiJuice doing?
If it is on continuously then it is not in low power mode (it should be blinking slowly and very weak).

A380Coding, 2022-04-05:
System Event Low charge: SYS_FUNC_HALT_POW_OFF, Raspberry Pi is being shut down (red power led on pi is off)
LED on pijuice is blinking green which should be ok.

A380Coding, 2022-04-05:
Yes I do, with a 60 second delay. And after shutdown it takes about 60 seconds until the red power led from the pi is out. Still thinking what else to test to narrow it down.

tvoverbeek, 2022-04-05:
What is connected to the HDMI port? Could that sip power?

A380Coding, 2022-04-05:
It is an old TFT display connected via an HDMI to DVI adapter. I can not imagine this to be the issue, but have not yet tried to disconnect after shutdown and connect before next startup to see if that has an impact on battery level. Since there is nothing else to do, I will check that next.

A380Coding, 2022-05-22:
I detected, that the Pi was running while the power was disconnected. It shut down but somehow restarted.
Could it be that pijuice starts the Pi since the battery is full and above the threshold even though there is no power connected?

A380Coding, 2022-05-22:
I just checked again and it seems to be the issue. The battery is full and the pi shuts down via cron. After shutdown also pijuice halts. Everything fine. After few minutes the power is disconnected. After some time the raspberry pi boots up on pijuice battery without the power connected (battery level is above threshold).
Is it possible to change the settings that reboot only occurs once power is connected and battery is above threshold?

A380Coding, 2022-05-22:
Is it right to have "Restore" active?

tvoverbeek, 2022-05-22:
Is your display powered externally?

A380Coding, 2022-05-22:
Yes it is. It has a separate AC power supply.

tvoverbeek, 2022-05-22:
You wrote
`After 60 seconds power to raspberry pi is off (red light on raspberry goes off).
10 minutes after that the USB power gets disconnected automatically (on purpose based on power consumption).`
If after shutdown by 'Software Halt Power Off', then due to external power being present for at least 10 more minutes the Pi will restart since there is power to charge the battery and the battery is above threshold. It will continue to run until the low battery threshold is reached.
Is this what happens?
Also I do not know if your externally powered TFT some how delivers 5V to the RPI via the hdmi-dvi adapter. Check if removing the display causes the same problem.

Anyway, the wakeup-on-charge  only works if power is connected to the charging circuit and the battery is above threshold.

A380Coding, 2022-05-22:
Interesting. The AC power will be turned of after 10 minutes below threshold (threshold here is the standby power of pi and display). I will set this to a lower value to be sure.
I did not count the time, could be 10 minutes, but not sure since I do not know what happens I was not expecting this. It should not reapear if the AC power (which is also the usb power) will be cut after let's say 5 minutes.

If above does not help I will check the 5V topic, but I doubt this. Since it is DVI it has only data lines available. And I saw the pi booting while no power is available, so it is obvious that this empties the battery.

While typing this. I was providing a shutdown command to the pi and let it (AC power permanently connected). It booted again after a bit over 1 minute. Is this 10 minute value configurable somewhere?

A380Coding, 2022-05-23:
Today the raspberry booted up again even though the USB power / AC get's shut down after 5 minutes now.
Can I choose another setting to reach the wanted behavior?

tvoverbeek, 2022-05-23:
As long as your external power stays on after the shutdown and the battery is above threshold the Pi will reboot.
In case the battery is above threshold at shutdown you need to disable wakeup-on- charge to avoid this.
To wake up the Pi again use e.g. the RTC.

A380Coding, 2022-07-02:
If the power is disconnected from Pijuice and the battery level is higher than the wake up on charge level, the raspberry will be started on battery. In case the no power option is used, it will be shut down immediately and be stuck in that loop until battery level drops below wakup on charge level.
It does not consider if power is connected/available or not.