PiSupply/PiJuice #854: Battery not being detected by pijuice_cli, and not charging

Peytoncl, 2022-04-04:
Im pretty sure i took the plastic covering off, unless theres more somewhere that i dont see? It flashed blue when i plug it in but then turns red and stays red, and says theres no battery connected.

Peytoncl, 2022-04-04:
I am using the pijuice battery that came with my purchase

tvoverbeek, 2022-04-04:
One thing to check in case your external power is connected to the Pi:
Is "GPIO input enabled checked? (On tha  Hat general tab in the GUi/ "General settings" tab in the CLI).
If not enabled the battery will not charge.

Peytoncl, 2022-04-05:
Yes it says that the gpio input is enabled and charging is enabled

Peytoncl, 2022-04-05:
i believe i have my battery connected, is theres something im missing? i just put my battery into the hat and dont do anythign else, should i ?

johnathongotcher90, 2022-04-05:
my computer only works when plugged in and it wasnt like this when i got it