PiSupply/PiJuice #856: Empty Battery - PiJuice stops working?

track02, 2022-04-13:
Hi,

We have a few units (RPi4) configured to use Pi Juice Zero boards to assist with start up and clean shutdown via button press/release

Power is supplied to the Pi Juice via the 4.2-10v terminal and the Pi Juice is configured to start up when a press is detected on J5

Several units were setup last year and have not been powered on for many weeks/months and as such the batteries have drained

When applying power to these units some of Pi Juice boards no longer activate and the battery doesn't charge, power is still supplied to the pi (assuming via gpio) but removing power will instantly shutdown the boards which isn't ideal

Strangely some boards seem fine and will start charging the empty battery

From initial testing it seems that swapping the drained battery out for one already charged seem to 'restart' the Pi Juice but this isn't too practical as the units are generally sealed in an enclosure

Would you be able to advise?

Many thanks

track02, 2022-07-20:
It's been a while and didn't hear back

We've started looking further into this, and it seems to be an issue when the PiJuice isn't powered 'directly' via micro usb and the battery completely drains, i.e. no status LED blinks

On dead units it seems that swapping to the micro usb 5v supply 'reboots' the PiJuice and it starts managing the battery again, even if the battery was fully drained, you can then swap back to 4.2-10v 

It's like the PiJuice gets 'stuck' in a state that it can't recover from when the battery fully depletes unless its powered directly. 

Assuming as there's no charge in the battery there is no power available for the PiJuice to perform an 'initial' boot and it's unable to tap off the 4.2-10v terminal it needs direct 5v - I may be way off so do correct if wrong

Difficult to test and fully confirm things as we need to wait for batteries to fully drain whilst connected to PiJuice but we should have another few ready to test this week or early next week

Would you have any thoughts or suggestions?

tvoverbeek, 2022-07-20:
When the battery completely drains the PIJuice MCU goes off.
If you reapply power it should start again. Obviously it does since the PiJuice voltage converters are able to supply the Pi in that case (as described in your April post).
It could be that it believes there is no battery connected. In that case charging will not happen.
What happens if you enable the 'no battery power on' setting?

track02, 2022-07-20:
Thanks for the reply

The PiJuice seems completely unavailable when it's in this 'stuck' state after battery drains, the power converters still work as underlying Pi will boot up

But trying to query a PiJuice itself via pijuice-cli would return errors and there was no entry in the i2c table, once a PiJuice recovers after micro USB supply is used it is again visible

I'll double check the 'no battery power' setting once batteries have drained again and we can continue testing
But if I remember from few weeks ago I think we did try setting it and it made no difference, as mentioned the only thing we found was to power via micro USB to 'reset' the PiJuice