PiSupply/PiJuice #857: Charging with a mobile phone charger

hlawati14, 2022-04-15:
Hello,
I just received my PiJuice and attached it to an Rpi 4.
I have the Rpi power supply 3V, 5A (usb-c), but I want to be able to plug the pi into charge with out it turning ON.
This obligates me to use the micro USB input of the piJuice.
I wonder whether its possible to to charge the micro USB of mthe piJuice using a normal mobile phone charger?
Will a Power Delivery fast charger work or it will fry up everything?

Thanks

tvoverbeek, 2022-04-15:
The micro-usb of the PiJuice does not support Power Delivery. As far as I know a PD charger falls back to normal US charging at 5V. You could also use a non-PD micro-usb phone charger

Stevo52, 2022-04-16:
I am a bit confused by what you say @tvoverbeek .....
When my PiJuice was purchased, I received a USB C (F) to USB C (M) and microUSB (M) adapter. That allows you to connect the RPi charger to both the RPi and the PiJuice. I juct connect the power supply to the PiJuice micro USB and it works? The documentation recommends this, with the RPI connected via the GPIO? I also connect a 7" Touchscreen with power to it via the GPIO (I actually use the PiJuice P3 connector for the 5V and Gnd).

hlawati14, 2022-04-16:
> The micro-usb of the PiJuice does not support Power Delivery. As far as I know a PD charger falls back to normal USB charging at 5V. You could also use a non-PD micro-usb phone charger

Thank you for the fast reply!
I will try using the PiJuice with my phone charger.
Are there any recommended ratings of voltage and current for the mobile phone charger to use? 
Will any Samsung normal charger work?