PiSupply/PiJuice #862: Rasp PI will not switch on when battery is charged

mauriceatkinson, 2022-05-04:
I have my Pi Juice connected with a battery and a solar panel so testing to see if working off grid is possible.  The pijuice turns the RPI off when the battery reaches the configured minimum charge setting.  However the pijuice does not switch the RPI back on when it reaches the wake upon charge setting.

I am obviously doing something wrong but the config is so simple nd I have read the wiki etc and still cannot find the solution.  Can anybody help?

tvoverbeek, 2022-05-04:
Which version of the software and firmware are you using?
In previous firmware versions the wakeup-on-charge setting is cleared when the Pi is woken up.
You would have to resend it just before shutdown. In most cases the pijuice service takes care of it, but not in all.
The latest firmware has an option to store the wakeup-on-charge setting permanently in the PiJuice non-volatile memory.

mauriceatkinson, 2022-05-04:
I am running software version 1.8 and firmware version 1.6.  I have looked in the sw settings but cannot find anything that relates to non-vol memory so would appreciate some direction here.  thanks

tvoverbeek, 2022-05-04:
You are on the latest software and firmware version.
It is the 'Restore' option on the System Task tab in the GUI or the System Task panel in the CLI

mauriceatkinson, 2022-05-04:
fantastic I will try this out.  Thanks very much