PiSupply/PiJuice #863: Pijuice Solar Panel turns itself off

Astronoms, 2022-05-05:
I am working on off-grid nature camera project and rely on solar power and Pijuice 12 000mAh batterry.
There is strange behavior when temperatures drop below 10 C during the night.
I rely that PiJuice itself controls the temperature and will not charge the battery  at low temperatures to avoid damage.
But now recently I discovered that in the morning after cold night Solar panel itself is off - the red LED is off. 
And to get it on again, I have to fold the panel for 8 seconds, then when exposing it again to light, LED turn on and everything works again. I tried the 22 W panel, the same behavior. And left panel disconnected from PI, still LED on the solar panel is off in the morning. 
I will check how it works then temperatures do not fall below 10C. But since I need to system to run also below 10C during few days in spring, I need to understand what causes panel to turn off.
Is there some temperature protection build in solar panel controller?
Regards, 
Andis

darkcry, 2022-09-29:
Same issue here. My setup is at home where it is at least 20°C.
PiJuice Solar Panel 22W unplugged is showing red LED indicator as expeceted when exposed to light. 
Then in the night in will turn off as there is not enough or no sunlight. Also expected. 
But then the next morning and during the day it will still not show the red LED again an thus no output is available.

This behavior is with or without anything plugged in. So it seams to be a Problem of the Panel itself or the power controller on it. Anyway to bypass it?

ChristopherRush, 2022-10-13:
Looking into this. My guess is that the solar panel gets itself into a state where is cannot recover, most likely due to the threshold of power required where the sun would dip in and out repeatedly, that in combination with the low temperature.

Schnibbles, 2023-04-08:
Hi folks,

@Astronoms @darkcry 

I've been experiencing the same issue since back in 2021, when I thought I was just going mad. Thankfully, this issue and [this blog post](http://blog.rfox.eu/en/Hardware/SolarPi_experiment_1_The_PiJuice_Fiasco.html) have convinced me it's not just my own inexperience. 

As a result, after checking the [datasheet for the Pijuice charge IC](https://www.ti.com/lit/ds/symlink/bq24160.pdf) and establishing that the PiJuice's J4 VIN input can handle up to 10V, I've checked if the solar panel output regulator is the problem. 

So as an experiment I:
- Pulled off the black box on the back of the solar panel and cut off the output regulator, leaving bare negative and positive wires
- Checked the open circuit voltage of the solar panel with a multimeter; for my 12W version it is around 7V
- Cut both ends off a 3m USB cable I had hanging around and wired it to the solar panel with a couple of wagos and very badly soldered it to J4
- Put the solar panel in direct sunlight all day with the Pi on, then shut down the Pi at 8pm and scheduled a wakeup for 8am

That was yesterday, and I've left it in the sun yesterday and overnight (in the UK, with temperatures below 10 degrees) and today the solar panel is charging the battery perfectly, exactly as it should be. Note that this has never worked for me before; it has always failed to charge the following day until physically closed and reopened.

The PiJuice status output is a little confused and thinks my Pi Zero is drawing 3A, but everything seems to function correctly.

I'll leave it for longer to make sure it's working consistently, but it does appear it's the output regulator for the solar panel that's causing the problem. If you're up for some electronics work, it should hopefully work for you too. 

If you have a larger wattage solar panel, your open circuit voltage may be higher than 10.5V, in which case you will need to wire in a buck converter. The over voltage protection of the PiJuice's IC goes up to 20V, so you will _probably_ not destroy the Pi either way. 

I was worried about heat issues with the PiJuice handling voltage conversion, but even my Pijuice in a sealed plastic wildlife cam case had no issues. 

I'm torn between annoyance that the PiJuice team sold me a paperweight for £60, and recognition that this is a very (useful!) niche product with a very niche market that can't have made a lot of money.