PiSupply/PiJuice #864: Turning on RPi without pressing the Pijuice On/Off switch?

ankitregmi, 2022-05-05:
When the battery power dies (no sun to recharge battery through solar panel) and RPi and PiJuice both lose power. How can we power the RPi without pressing the ON switch in the PiJuice when the battery starts charging again?

mmilann, 2022-05-05:
Setup "Wakeup on charge" (Trigger level - 0%) with "Restore" option enabled and also enable "No battery turn on" option from General menu.

ankitregmi, 2022-05-06:
Thanks I will try that and see how it works.

johansmitsnl, 2022-12-06:
It worked for me. Minimum is 1% you can set. But there seems a delay when the power is put back in before it starts. This delay is this possible to be shorter?

tvoverbeek, 2022-12-06:
Sorry, not really. The firmware first has to start and determine the charge level before it turns on the 5V to the RPi.

johansmitsnl, 2022-12-06:
@tvoverbeek thanks for the quick response