PiSupply/PiJuice #866: Issues when connected with cable

funkyfourier, 2022-05-10:
My PiJuice with the 5000 mAh battery is working great when connected directly to the Pi using the pin socket. However, when I try using a cable, either a flat cable or individual jumper cables, I get some weird behaviour.

At first the Pi would start booting but would shut down about when the splash screen for Raspberry OS appears. Then I switched Power regulator mode to DCDC which makes it boot up, but the battery icon is not displaying and when I go to PiJuice Settings it says that it cannot connect to the I2C bus.

I have checked for continuation on each of the pins used (2, 3, 5, 6), and they have contact between the female connector on the bottom and the male pins on top. Also I checked continuation between the GPIO pins on the RPi and the corresponding pins on the PiJuice while connected with the flat cable and the individual jumper cables.

Thanks for any suggestions.

Connected using jumper cables:

[image]

I2C error message:

[image]

Flat cable from Adafruit:

[image]

funkyfourier, 2022-05-10:
Thanks @tvoverbeek. It works when connecting the PiJuice directly on the pins on the Pi4. I have now tried the following:

* Connect the other 5V pin as you suggested
* Connect the 3.3V pin
* Connect 3 of the other ground pins
* Connect all the pins (using the flat/ribbon cable)
* Power the screen with a seperate USB cable/wall adapter
* Have USB power plugged in on the Pi or the PiJuice

Also when powering up using SW1 the LED goes immediately green but some time during the boot process it switches off and starts blinking as it does when not connected to anything.

When connected with cables (and not working correctly) I get `Could not open file '/dev/i2c/1'` when running `i2cdetect -y 1`. Have not tried `i2cdetect -y 1` when connected directly on the pins, but in that case the PiJuice Settings and battery icon works perfectly.

So the diagnose as it seems to me is that i2c is not working when connecting through cables. Does that sound correct? If so, is there any reason it should not work as I have set it up?

tvoverbeek, 2022-05-10:
This is probably what happens:
With the PiJuice connected without cables the HAT eeprom is read at boot and enables  and loads the RTC OS driver (hence the 'UU' at 0x68).
With Pijuice connected via cables the HAT eeprom is not read.
It seems i2c is not enabled by default.

Without the PiJuice connected run raspi-config (`sudo raspi-config`-> Interface Options -> I2C -> Yes).
After this `ls -l /dev/i2c-1` should show something like: `crw-rw---- 1 root i2c 89, 1 Apr 26 18:57 /dev/i2c-1`.
This assumes you are running Raspberry Pi OS and are logged in as the default user.
If this works connect the PiJuice with cables and test again.

funkyfourier, 2022-05-10:
Heck yeah! It works!

It was stupid of me not to check that I2C was enabled, I assumed that it was since it worked when not using the cable. 

Thanks a bunch for excellent and swift support.

closed 2022-05-10 by funkyfourier

tvoverbeek, 2022-05-10:
@funkyfourier Glad it is working now.