PiSupply/PiJuice #868: Software Halt Power Off does not seem to be working

funkyfourier, 2022-05-19:
The option "Software Halt Power Off" does not seem to do anything on my setup. When powering off via the main menu "Logout"->"Shutdown", or running `sudo halt` / `sudo poweroff`, the red light on the RPi stays on several minutes after. In this mode the only way to power on again is to hold SW1 for 20 sec until it cuts the power to the RPi completely and then press it again.

"Software Halt Power Off" is set to 30 seconds and "System task enabled" is checked. It is a RPi 4B running Raspbian GNU/Linux 11 (bullseye) and PiJuice firmware V1.6_2021_09_10.

`rpi-eeprom-config` tells me:

BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

Should not the power to the RPi be cut off after 30 seconds regardless of shut down method?

tvoverbeek, 2022-05-19:
Just to be sure, "Software Halt Power Off" only works when running on battery only.
It does not work when you are powering via the Pi4 USB-C.

Nadahar, 2022-06-13:
> Just to be sure, "Software Halt Power Off" only works when running on battery only.
> It does not work when you are powering via the Pi4 USB-C.

It works for me when powered via the micro-USB on the PiJuice as well (not just on battery). It obviously can't work if the RPi is "powered directly" (GPIO or USB-C).

tvoverbeek, 2022-06-13:
@Nadahar Correct."Software Halt Power Off" also works when external power is connected to the PiJuice micro-USB.

funkyfourier, 2022-08-18:
OK, so I got a new PiJuice **Zero** and I am still having similar issues, although not quite identical. The PiJuice Zero is hooked up to a RPi4B with a stacking header. It is running FW 1.4 as it was out of the box, and the RPi4B is running bullseye with the latest updates installed. The PiJuice Zero has the official 5000mAh battery connected.

"Software Halt Power Off" is set to 20 and "System task enabled" is enabled. No other options have been changed.

What I am getting now is this:

* When I turn the machine off from "Logout"->"Shut Down" the power to the RPi4B cuts off after approx. 20 seconds.
* When I turn the machine off with `sudo halt` the power to the RPi4B cuts off after approx. 20 seconds.
* When I turn the machine off by pressing SW1 for ten seconds it appears to do the same as above, and I get the black "plymouth-poweroff.service", but the power to the RPi4B is not cut off after several minutes.

These symptoms are consistent with or without the micro USB plugged into the PiJuize Zero.

**Note:** In none of these cases has the USB-C plug on the RPi4B been connected to anything.

Any feedback would be appreciated. I am building a single-app portable device with a touch screen, so the ability to proper power off via the button would be a highly useful feature.

Will also test with the newest firmware and update below.

tvoverbeek, 2022-08-18:
The "Software Halt Power Off" setting only applies to shutdown/halt initiated by user software (your first 2 bullets).
The SW1 long press effect is determined by the function you have assigned to SW1 LONG_PRESS_1
SYS_FUNC_HALT wil shut down the system but keep the 5V on (probably the case for your 3rd bullet).
Change it to SYS_FUNC_HALT_POW_OFF. The power off will occur after ca 60 sec. This value is har-coded.
See lines 66-69 in pijuice_sys.py (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py)
This is all documented in the SOftware README on github.