PiSupply/PiJuice #869: GPIO Power input NOT PRESENT

proggprod, 2022-05-22:
Hi,

Having trouble getting power from GPIO power input.
This PiJuice was working on a Pi 4. I am now moving it to a Pi 3 and this issue has presented itself.
I have pijuice cli installed and the Status page does show a voltage and current for GPIO, but also 'NOT_PRESENT'.
I have a micro USB plugged into the pijuice for now to keep the battery charged, but even that shows as 'WEAK', even though it should be a good power supply.

Any suggestions?
 
[image]

tvoverbeek, 2022-05-22:
Your HAT status hows an (estimated) current of 1.347A onthe GPIO.
Positive current means current going from the PiJuice to the RPi.
When power is connected to the RPI, "USB Micro power input" would show NOT_PRESENT and
the current for 'GPIO power input' would be negative if charging.
Mos ot the power in your case seems to be going to the Pi3, so there is not enough reserve to charge
the battery properly, hence te 'WEAK'.
What peripherals are connected?

proggprod, 2022-05-22:
Thanks for the info.
Now that the battery is charged, the micro USB input says PRESENT.
As it is hovering around 100%, Battery changes momentarily from CHARGING_FROM_IN to NORMAL, and GPIO momentarily changes from NOT_PRESENT to PRESENT. But not both at the same time, and not for very long.

I have a POE HAT (not being powered by POE yet),  and a PiJuice, no other peripherals.

tvoverbeek, 2022-05-22:
Does it behave the same when the POE hat is not connected?

proggprod, 2022-05-22:
Hi,

Yes. Same without POE HAT. But, I've plugged it to POE power now (didn't have it set up before now), and now it shows GPIO as PRESENT.

Sorry for wasting your time. This power supply must not be as good as I thought. It has always powered my Pi 3 just fine and came with it when I bought it. Maybe something has aged in it recently.