PiSupply/PiJuice #873: Pijuice Zero + battery = shut down after reboot

SneeuwvosNL, 2022-06-04:
I have my Pijuice installed on my Raspberry Pi Zero 2 W.
I have the latest Pijuice firmware: 1.6
No battery attached.

**At config I have:**

_[General settings]_
I2C address: 14
I2C address RTC: 68
[X] GPIO input enable
[ ] EEPROM write unprotect
[ ] Charging enabled    (FYI: battery is charged for >90%)
[X] No battery turn on

_[System Task]_
[X] System task enabled
[ ] Watchdog [ ] Restore
[ ] Wakeup on charge [ ] Restore
[ ] Min charge
[ ] Min batery voltage 3.3
[X] Software Halt Power off Delay period [seconds]: 30

_[Battery settings]_
Profile: PJLIPO_500
[ ] Custom

_[System Events]_
[ ] (All settings disabled)

At "Wakeup Alarm" I have setup a schedule. I turn off the Raspberry Pi with "sudo halt". So far so good, it's boots up right on time!

FYI: I only have the USB cable attached to the Pijuice Zero. Whenever I pull it out + push back it the Raspberry Pi boots. Which is good so far again!

**----- now the problem part ------**

Whenever I connect a PiJuice Zero 500mAh Battery to the Pijuice, the Raspberry Pi starts up. That's fine.
But as soon as the boot has completed, the Raspberry Pi immediately shuts down.

I've been struggling with the config for hours, tried many different settings, but I can't find a solution why this happens. I hope someone can help me.

tvoverbeek, 2022-06-04:
Questions:
Is the pijuice service running? CHeck the output of `systemctl status pijuice.service`
When you connect the 500mAh battery is the external power still connected to the PiJuice Zero?
Does it then start up immediately after you insert the battery or at the scheduled start-up time?

SneeuwvosNL, 2022-06-04:
Thanks for helping me out so quickly!

_Is the pijuice service running?_: Yes, it says it's running.
_When you connect the 500mAh battery is the external power still connected to the PiJuice Zero?_: No, I only plugged in the battery (No Micro USB connected to the Rpi nor Pij).
_Does it then start up immediately after you insert the battery or at the scheduled start-up time?_: It starts immediately after I insert the battery (again: without Micro USB connected to the Rpi nor Pij)

tvoverbeek, 2022-06-04:
At the moment I have no explanation.
When I try this with a PiJuice HAT on a RPI4 with your settings and no battery and no external power connected, inserting the battery does not start the Pi.
I'll try to reproduce your scenario using a Pi Zero 2W and a PiJuiice Zero with the 500 mAh batttery.
Stay tuned. (Later meer ...)

SneeuwvosNL, 2022-06-05:
I am looking forward to your findings.

I purchased another Raspberry Pi Zero 2 W which comes pre-soldered (so: WH version to be precise). So I can test it with that one too as soon as I receive it somewhere the coming week. I only ran tests with my own soldered Rpi Zero 2 W. Could it be a soldering issue?

tvoverbeek, 2022-06-05:
I tested today with a Pi Zero 2W and Pijuice Zero with 500 mAh battery
When I connect the battery (no external power connected) the RPi does **not** switch on.
Only the PiJuice Zero Led lights up shortly (green, since battery was charged) and then changes to a slow weak blink
(= PiJuice Zero in low power state). I can then power on the Pi using a key press.

However I noticed the battery connector pins stick out on the bottom side of the PiJuice Zero.
If you press the PiJuice Zero on the Rpi the pins might make contact with the Pi Zero 2 W.
To eliminate that you could try to put the PiJuice Zero on a 40-pin extension header. and see of the probllem goes away.

SneeuwvosNL, 2022-06-06:
Thank you for checking this out.

I am using a 40-pin extension header. I also tested with different extension (/stack) headers. Just to make sure that wasn't it either. 

I am using a stack header because I want a display connected on top of the Pijuice, but I haven't during testing.

Coming week my new Raspberry Pi Zero 2 W(H) will arrive and tell you my new findings. I want to make sure if it's my soldering (on my current Rpi) or not. 

To be continued :-)

SneeuwvosNL, 2022-06-06:
I couldn't help myself just to keep playing with the device to see if I could find a fix. I think I spend over 5 hours in total on it now, haha. I even (re)installed the latest firmware 1.6 onto it. No help so far.

**I found something else interesting though...**

Whenever I typed `sudo systemctl status pijuice.service` command it said the service was running. Which is good.

So I disabled its service by typing: `sudo systemctl disable pijuice.service`

I rebooted the system (battery plugged in, no auto start = OK. I plugged the micro usb in, no auto start = OK) so far so good.
I pressed the "SW1"-button and it booted. Which is good too.

**It booted AND IT DID NOT SHUT DOWN.**
Normally when I did this the led (left side of SW1 button) started blinking red and shut the system down.

So my conclusion so far is... why does the service shut down my Rpi at shutdown.