PiSupply/PiJuice #874: PiJuice seems to be incompatible with Python3.10

GerBreOwn, 2022-06-09:
I have an application that has been running fine until I upgraded Python3.9 to 3.10. Now when I run the application I get the error: "ModuleNotFoundError: No module named 'pijuice'".  If I run "sudo apt install pijuice-base" I get the message "pijuice-base is already the newest version (1.8)." If I run pijuice-cli it runs as advertised.

Any idea why it does NOT run in my application and how it can be fixed?

Thanks

tvoverbeek, 2022-06-09:
Is /usr/lib/python3/dist-paclages in your Python path?
The pijuice-base install installs pijuice.py in /usr/lib/python3.9/dist-packages but also a symbolic link in /usr/lib/python3/dist-packages.
```
tvo@kaifatest:/usr/lib/python3/dist-packages $ ls -l pijuice.py
lrwxrwxrwx 1 root root 43 May 26 14:15 pijuice.py -> /usr/lib/python3.9/dist-packages/pijuice.py
tvo@kaifatest:/usr/lib/python3/dist-packages $
```
Brute force way is to copy the pijuice.py module to /usr/lib/python3.10/dist-packages

GerBreOwn, 2022-06-10:
I did not have a directory "/dist-packages" under python3.10 so I copied the one from 3.9. That still did not work. I then created the PYTHONPATH environment variable pointing to "/usr/lib/python3/dist-packages". That got rid of the "no pijuice module error".

How ever it now complains about "No module named 'smbus'". How do I fix that?

tvoverbeek, 2022-06-10:
If you had 3.9 installed then it is a similar problem.
All the python packages in the raspberry pi repositories are for 3.9.
smbus (package python3-smbus) is not written in Python but is a C shared library callable from python.
The 3.9 version probably does not work with 3.10. Moreover if you are on a 64-bit system you need a specific 64-bit version.
Suggest you look at smbus2 on PyPi which is a pure Python implementation.

GerBreOwn, 2022-10-21:
The problem has raised its ugly head once again. "**ModuleNotFoundError: No module named 'pijuice'". If I run "sudo apt install pijuice-base" I get the message "pijuice-base is already the newest version (1.8).** It was running several months ago about 15 times and then it quit. I haven't been checking that program as I have been working on other things.

GerBreOwn, 2022-10-21:
I just did "sudo systemctl status pijuice" and get the "ModuleNotFoundError"

tvoverbeek, 2022-10-21:
What does your python path look like?
```
import sys
print(sys.path)
```

GerBreOwn, 2022-10-22:
I just did "sudo systemctl status pijuice" and get the "ModuleNotFoundError"

> What does your python path look like? echo $PYTHONPATH
/usr/lib/python3/dist-packages:/home/gerald/.local/lib/python3.10/site-packages

> 
> ```
> import sys
> print(sys.path)
> ```
['', '/usr/lib/python3/dist-packages', '/home/gerald/.local/lib/python3.10/site-packages', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/usr/lib/python3.10/site-packages', '/usr/lib/python3.10/site-packages/spidev-3.5-py3.10-linux-armv6l.egg', '/usr/lib/python3.10/site-packages/myDevices-2.0.4-py3.10.egg', '/usr/lib/python3.10/site-packages/paho_mqtt-1.6.1-py3.10.egg', '/usr/lib/python3.10/site-packages/requests-2.28.1-py3.10.egg', '/usr/lib/python3.10/site-packages/psutil-5.9.2-py3.10-linux-armv6l.egg', '/usr/lib/python3.10/site-packages/netifaces-0.11.0-py3.10-linux-armv6l.egg', '/usr/lib/python3.10/site-packages/enum34-1.1.10-py3.10.egg', '/usr/lib/python3.10/site-packages/urllib3-1.26.12-py3.10.egg', '/usr/lib/python3.10/site-packages/idna-3.4-py3.10.egg', '/usr/lib/python3.10/site-packages/charset_normalizer-2.1.1-py3.10.egg', '/usr/lib/python3.10/site-packages/certifi-2022.9.14-py3.10.egg']

tvoverbeek, 2022-10-22:
You have `/usr/lib/python3/dist-packages` in your python path.
The pijuice-base install puts pijuice.py in `/usr/lib/python3.9/dist-packages` and a symbolic link to pijuice.py in `/usr/lib/python3/dist-packages` so all 3.x versions should find it when `/usr/lib/python3/dist-packages` is in the python path.
What error message(s) do you get if you run  the `import pijuice` command in the python interpreter?
Also can you check if pijuice.py is in `/ust/lib/python3.9/dist-packages` and the symbolic link in `/usr/lib/python3/dist-packages`?