PiSupply/PiJuice #876: Suggestion: Add option to detect shutdown via GPIO 3.3V

Nadahar, 2022-06-13:
The `Software Halt Power Off` configuration option and the `SYS_FUNC_HALT_POW_OFF` and `SYS_FUNC_SYS_OFF_HALT`(?) rely on timers to let the OS shut down gracefully before cutting the 5V. The former is configurable, the others are hard-coded to 60 seconds if I understand the documentation correctly. 60 seconds might be a bit short though, I timed my (pretty much unconfigured/taskless) headless openHABian installation on a RPi 4 4Gb to 44 seconds from running `shutdown -h now` until the 3.3V was cut. It's not hard to imagine that with some actual work, a GUI, applications that doesn't respond immediately etc 60 seconds would easily be exceeded. Ideally they would all respect the same setting, if that's not possible (software/firmware divide) maybe there should be a similar configuration option for the "firmware" part.

I don't know if the MCU is connected to/can sense the RPi GPIO 3.3V, but if it is/can, it should be possible to have an option to use the 3.3V as a "cue" instead of using the timer. Users that wanted to use this would have to configure `WAKE_ON_GPIO=0` and `POWER_OFF_ON_HALT=0` with `rpi-eeprom-config`. This makes the 3.3V turn off (at least on the RPi 4) when the system has actually shut down, regardless of how long it takes. Alternatively/in addition some other GPIO pin could also be used to trigger this I guess, although that might still need some delay as I assume there's not control in the last parts of the "shutdown procedure".

File system corruption is something I really want to avoid, and as it is now, with the hard-coded 60 seconds limit when shutting down due to low battery, this might be very difficult to protect against.

tvoverbeek, 2022-06-17:
As far as I know the RPi 3.3V GPIO pins are not used/connected by the PiJuice.

If the hardcoded 60 sec for SYS_FUNC_HALT_POW_OFF (line 68 in https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py) is not long enough you could do the following:
The 60-sec timer is started when pijuice_sys.py executes SYS_FUNC_HALT_POW_OFF before it calls `sudo halt`.
The OS shutdown stops all services in reverse order. When it stops the pijuice service pijuice_sys.py is called again with a 'stop' argument. The code after line 327 in pijuice_sys.py is then executed.
First it calls the user function associated with the 'system stop' event ('system start' and 'system stop' events were introduced in the 1.8 version of the software). In this function you could reprogram the timer so it suits your shutdown sequence.

Nadahar, 2022-06-26:
@tvoverbeek Thank you for the reply.

For me, the most important part of having a "UPS solution" is to unsure safe shutdown without filesystem corruption (the use of zram makes the chance of something going wrong even higher). The added runtime for shorter power outages is a bonus of course, but the most important thing is that I can trust that the system will still start up and work. My worry isn't so much that 60 seconds isn't enough - it seems to be more than enough normally. But, I know how little it takes for some software to have to wait for a timeout for the "shutdown time" to change drastically. There are many possible situations where this can happen on a non-regular basis, maybe especially during a power loss where network connectivity or other devices might suddenly be unavailable. That's why I worry that 60 seconds might not do, and that it's very hard to predict what is a reasonable time to wait. You don't want to drain the battery more than necessary by setting a very long delay "to be safe" either, and it might end up shutting down because the battery runs out in the meanwhile instead.

This is why I thought of the 3.3 V as a possibility to circumvent the whole problem, by the time that shuts down the OS definitely has stopped. But, since that's not a possibility, I'd say the second best thing is a delay that starts "counting" as late as possible in the shutdown process.

I haven't really used Python since 2005, and I'm not familiar with what happens on the "firmware side", so it's a bit hard for me to get an overview. But, from what I think I understand, I see a design that I'm not sure why is the way it is. According to your explanation, `pijuice_sys.py` is called with `stop` when the `pijuice` service shuts down, which should normally be reasonably late in the shutdown process..? That looks to me like the "best" place to initiate the power cut timer - as opposed to at the beginning of the shutdown process.

As far as I can tell, when the system isn't shutdown by `pijuice` the `HALT_FILE` is never created, and the delay is initiated at this "late stage" where most of the potential events that can delay the shutdown has already happened. But, if the system shutdown is initiated by `pijuice` via `SYS_FUNC_HALT`, `SYS_FUNC_HALT_POW_OFF` or `SYS_FUNC_SYS_OFF_HALT`, the `HALT_FILE` is created at the beginning of the shutdown process which signals to `pijuice_sys.py` to *not* initiate the power off timer at the late stage. If `SYS_FUNC_HALT_POW_OFF` is executed, the power off timer is initiated at the very beginning of the process instead, making the 60 second delay potentially destructive to the file system.

The only way I can find to make this make sense, is that there are users who need `SYS_FUNC_HALT` and/or `SYS_FUNC_SYS_OFF_HALT` to NOT power off the 5V, while at the same time having "Software halt power off" enabled. My question is if the need for this combination is needed at all. I mean, why would you want the 5V to shut of when you shut down the system manually, but not when it is triggered by a "system event"? 

What is the benefit of leaving the 5V on in the first place? I would think that could only be beneficial if something else is also powered either by the 5V or the "SYS" output and you want that to keep running until the battery is completely drained (at which time it will lose power anyway).

I know how the history of a software development project sometimes can result in some seemingly illogical solutions because things have been changed a number of times, while trying to not break anything for existing users. I don't know if that's what have happened here, or if I'm missing some other scenario where this makes sense. But, to me it would seem both simpler and "better" to let "Software halt power off" handle the power off delay regardless of what triggered the shutdown. That way the timer would start as late as possible, the delay would be configurable and the code would be simpler.

As I understand it, "wake on GPIO" requires to 5V to be powered to work, but I didn't think that PiJuice utilized that? Except for that situation, I'd say it's usually better to have the 5V powered off, since it makes it easier to start the RPi again (the SW1 click doesn't do anything when the system is halted but the 5V is on). I also assume that the "Wake on charge" doesn't rely on the 5V being on? In fact, I'm wondering how it would handle the 5V being on, because that would require it to use the "wake on GPIO" function of the RPi to enable booting?

So, as you can see, I'm a bit confused, but I'm trying to get a grasp of the process. I'd appreciate any feedback you could give to better my understanding, all I want is to end up with a solution that I know is "safe" from "sudden power loss" in all forms. I assume that the same is true for most other PiJuice users.