PiSupply/PiJuice #878: Wakeup on charge should depend on ext. Power

phaupt, 2022-06-16:
Hello! In my scenario I want to use the system task as follows:
- "Min charge" threshold at 60%. If I remove external Power to the Pi, the PI should keep running until PiJuice battery is below 60%.
- Wakeup on charge at 0%. If external Power to the Pi is present, the Pi should be up and running.

The System events:
- "Low charge" = SYS_FUNC_HALT_POW_OFF

If PiJuice battery is 100% and then I remove the external Power to the Pi, PiJuice keeps the Pi running until it drops below 60%. This works as expected.

However, if external Power to the Pi is present AND the PiJuice battery below 60%, then it does not properly work. It wakes up the Pi and immediatly shutdown again. It runs in a boot/shutdown loop until the battery is charged above 60%.

Is it possible to use the "Min charge"/"Low charge" in a way that it also considers the presence of the external power? The PiJuice should only shutdown the Pi if battery is below 60% AND no external power is present (=not charging). Whenever it is charging, the Pi should be running. How can this be configured?

tvoverbeek, 2022-06-16:
Connect a user function to the "No Power" event.
This user function should enable the "Low Charge" event.
At boot/wakeup disable the "Low Charge" event when external power is present.

phaupt, 2022-06-16:
Raspberry Pi 4 with teslausb image 

Firmware 1.6 (up to date)
Reset all settings to factory default, then I've changed only these settings, using PiJuice CLI:

* Battery Profile BP7X_1820
* Charging enabled
* No buttery turn on

System Task:
[X] System task enabled
[X] Wakep on charge: Trigger level 1% [X] Restore
[X] Min charge: Threshold 60%
[X] Software Halt Power Delay period: 30

System Events:
[X] Low charge: SYS_FUNC_HALT_POW_OFF

pijuice.service is enabled, active and running.

If the battery is fully charged, it keeps the Pi running even if I remove USB-C/Power until it drops below 60%. If I reconnect the USB-C/Power while the battery is still below 60%, it will do the boot/shutdown cycle.

tvoverbeek, 2022-06-16:
Just ran a test on my Pi4/PiJuice with your settings and I do not get a boot loop when I apply external power with the charge level below the threshold.
How do you power the Pi4/Pijuice combination? Via the PiJuice microUSB or via the Pi4 USB-C?
I suspect some interaction with your teslausb version. It seems a heavily modified version from standard Raspberry Pi OS.