PiSupply/PiJuice #882: PiJuice failing to boot Raspberry Pi

phillmagic, 2022-06-27:
Hello, i just purchased a PiJuice for my Raspberry Pi 3. While using an extended 12000 mAh battery, fully charged, wont boot up. Stops at boot and shuts the Pi off completely, any help with this shall be appreciated. Spent alot of money on extra battery, 22 watt panel and pijuice

tvoverbeek, 2022-06-27:
Need more details. Both battery and Solar panel connected?
WHen powering the Pi3 with a normal power supply can you list your settings (either from cli or gui)?
But first: Did you  install the software and what OS are you using?

phillmagic, 2022-06-28:
Closing issue, sorted it out. Had to solder back the old battery holders, seems the ones we had put were not tight enough. Thanks otherwise

closed 2022-06-28 by phillmagic