PiSupply/PiJuice #891: Documentation for i2c interface

PhillippOhlandt, 2022-07-23:
Hello,

I looked through this repo and could not find the documentation for the i2c interface. I am porting the python lib to a different language but not every function in the python lib is documented (like `GetRsocEstimationConfig`) so I wanted to look directly into the i2c interface documentation. Also, any other documentation which might help me, would be good.

tvoverbeek, 2022-07-23:
The actual i2c commands you will haave to deduce from pijuice.py (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py).
Here is the snippet for `GetRsocEstimationConfig`:
```
    rsocEstimationOptions = ['AUTO_DETECT', 'DIRECT_BY_MCU']
    def GetRsocEstimationConfig(self):
        result = self.interface.ReadData(self.BATTERY_TEMP_SENSE_CONFIG_CMD, 1)
        if result['error'] != 'NO_ERROR':
            return result
        else:
            d = result['data']
            if ((d[0]&0x30)>>4) < len(self.rsocEstimationOptions):
                return {'data':self.rsocEstimationOptions[(d[0]&0x30)>>4], 'error':'NO_ERROR'}
            else:
                return {'error':'UNKNOWN_DATA'}
```
Elsewhere in pijuice.py you find ` BATTERY_TEMP_SENSE_CONFIG_CMD = 0x5D`.

Hope this helps with the porting.

PhillippOhlandt, 2022-07-23:
Ah ok, that's unfortunate. The only hint that this function has something to do with battery temperature sensing is the command name. I hoped there would be an in-depth documentation so some code parts from the python lib can be better understood. 

For example, when retrieving the battery temperature:

```python
    def GetBatteryTemperature(self):
        result = self.interface.ReadData(self.BATTERY_TEMPERATURE_CMD, 2)
        if result['error'] != 'NO_ERROR':
            return result
        else:
            d = result['data']
            temp = d[0]
            if (d[0] & (1 << 7)):
                temp = temp - (1 << 8)
            return {'data': temp, 'error': 'NO_ERROR'}
```

It is requesting two bytes (because the checksum is on the third byte) but only the first byte is used. And then there is this check:

```python
            if (d[0] & (1 << 7)):
                temp = temp - (1 << 8)
```
I don't know what it is doing.  If I just use the first byte without that check, I get e.g. `35` but when I implement that check, I end up with `-220`. 

Having this documented somewhere would have been nice.

tvoverbeek, 2022-07-23:
That is converting from unsigned to signed byte

PhillippOhlandt, 2022-07-23:
Ah, got it. Actually, my code was buggy and the if-condition was always executed, leading to above mentioned output which was obviously not correct.