PiSupply/PiJuice #893: RPI4 Not running on battery with PiJuice

JakeEllis, 2022-07-29:
Raspberry Pi 4
Raspbian 10
PiJuice settings are all set to the default (no system tasks or events)

**Issue**:
My pi runs perfectly fine when plugged in to a power source via USB-C or an external battery into the PiJuice micro-usb connection and I can see that the battery is charging and PiJuice settings appear to be reading everything correctly. However, when I remove the USB-C or micro-usb power source and run my pi with only a battery, the Pi tries to boot over and over again but never makes it to the home screen. I have tried with the battery that comes with the PiJuice as well as a new 12000mAh battery and they are both having the same issue.

I have attached some videos of what is happening. In these videos I am showing that I am charging the battery through the PiJuice with an external battery. When the PiJuice battery is at a good percentage I remove the external battery and it seems to work for a moment but then goes into a cycle is turning off and on.

The videos are too large, so here is a link to my Google Photos album.

https://photos.app.goo.gl/ioGShMDyx2ANHdeK8

tvoverbeek, 2022-08-02:
System Task enabled? Check with 'systemctl status pijuice'
Pi 4 is relatively power hungry and the default battery (BP7X) is barely able to supply sufficient power.
Any peripherals connected? (Did not see any in the videos).
Typically the Pi shuts down if it detects under-voltage.

JakeEllis, 2022-08-03:
No system tasks or system events enabled. I tried this with a 12000mAh battery as well but had the same issue. 

I have a SixFab LTE HAT that I have used but the issue happens whether that is installed or not. Like you mentioned, in the video, I had removed the LTE HAT and the issue persisted. 

the systemctl status pijuice command returns these results:

 pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2022-07-29 13:26:58 EDT; 4 days ago
 Main PID: 378 (pijuice_sys.py)
    Tasks: 1 (limit: 1431)
   CGroup: /system.slice/pijuice.service
           \u2514\u2500378 /usr/bin/python3 /usr/bin/pijuice_sys.py

Jul 29 13:26:58 raspberrypi systemd[1]: Started PiJuice status service.
Jul 29 13:27:03 raspberrypi pijuice_sys.py[378]: RTC os-support OK

JakeEllis, 2022-08-08:
@tvoverbeek following up. Do you have any other thoughts. Could the pins be dysfunctional on the pijuice?

tvoverbeek, 2022-08-09:
@JakeEllis No, I do not have any other suggestions. Pins being dysfunctional is very unlikely, unless they have been fried somehow.