PiSupply/PiJuice #894: RPi 3B+ not starting

PUW149, 2022-08-01:
I have a PiJuice with BP7X battery, software 1.8, firmware 1.6, on a RPi 3b+, with default config (checked via software).
Was working fine, but now SW1 not waking Pi even when BP7X fully charged. Long press of SW1 does power off Pi when running on mains. J5 or GPIO 5-6 do not power on the Pi either.

Not sure what other diagnostics to try?
Thanks, Simon

tvoverbeek, 2022-08-02:
Enable 'No battery turn on" and with external power connected to J5/PiJuice micro USB test if SW1 turns on the Pi.

PUW149, 2022-08-04:
Thanks, that worked ie. RPi started from SW1. Does that means the BP7X is a write-off? Doesn't start from battery even when battery is fully charged. The PiJuice App on the RPi doesn't show anything bizarre on the battery front, although the BP7X does seem to discharge quickly.

tvoverbeek, 2022-08-04:
Probably your BP7X is bad. I had to replace my firrst BP7X for the same reason (showed full, but could not deliver enough power). Remember these BP7X were originally made for the Motorola flip phones (Razr) of 18 years ago.

PUW149, 2022-08-05:
Thanks, I will get a new battery