PiSupply/PiJuice #895: RTC Wakeup causes Pi to reboot when already on

johncorbett65536, 2022-08-07:
Hi.
I have a Pi Zero W with PiJuice Zero configured with RTC Wakeup alarm as follows:
Wakeup enabled
Day: Every day
Hour: Every hour
Minutes period: 5 minute
Everything else is configured as factory defaults.
When the Pi is off, this causes the Pi to be powered on at 5 minute intervals as expected.
What is unexpected is that when the Pi is already on, it is forced to reboot at the 5 minute interval.
When PiJuice logging is enabled (python3 /usr/bin/pijuice_log.py --enable "OTHER|5VREG_ON|5VREG_OFF|WAKEUP_EVT|MCU_RESET") there is a WAKEUP_EVT triggered each time this happens with "TRIGGERS: RTC" (which is the same as when the Pi is off).

Q: Is this expected behaviour?
Q: If so, is there a recommendation for how to prevent reboots when power is already on?

Firmware version: PiJuice-V1.6_2021_09_10.elf.binary

tvoverbeek, 2022-08-07:
You wrote:
`Q: Is this expected behaviour?`
A: Yes. in case the Pi is shut down and the 5V to the Pi remains, the way to restart it is to interrupt the 5V supply.
This is what is happening.

`Q: If so, is there a recommendation for how to prevent reboots when power is already on?`
A: At the start of your run disable the EnableWakeUp and at the end before shutting down enable it again.
See https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-rtc-alarm the desciption of SetWakeupEnabled(status)

johncorbett65536, 2022-08-07:
Thanks @tvoverbeek.  Good to know.  I have a fallback strategy of setting a wakeup time for a specific time in the future at the start of the run (in case of crash or whatever) then setting normal regular wakeups again at the end of the run.
Great product by the way.