PiSupply/PiJuice #897: When raspberry is turned off - battery is still active

chrystelenicot, 2022-08-11:
Hello,

A quick noob question, 
I'm running my PiJuice on a 12.000 mAh battery.
However when I turn off the raspberry the battery is emptying it self instead of just be off.
Someone could help me where to change those parameters ?

thanks on advance !

tvoverbeek, 2022-08-11:
You have to make sure the 5v from the PiJuice to the RPi is turned off after shut down. Otherwise the battery will be drained by the residual power use of the RPi (RPi power led stays on).
WIth the Pijuice 5V turned off the PiJuice MCU will go to a low power mode (the RTC keeps running) and the battery will be able to keep it running for several weeks/months depending on battery capacity and charge level.

chrystelenicot, 2022-08-17:
Thanks for your help !  I switched it, it works great !

closed 2022-08-17 by chrystelenicot